TheZeroBeast/TZB-MMU2S-Firmware #6: Stopping randomly as if MMU2 missed a toolChange command from MK3

TheZeroBeast, 2018-11-05:


TheZeroBeast, 2018-11-05:
Master-Patch1 is being tested regarding this issue.

TheZeroBeast, 2018-11-08:
I think all aspects are covered now for ack on all commands to MMU2 apart from P0 as this can be done later if needed. Final testing before versioning and merging to master.

AbeFM, 2018-11-08:
Looking forward to trying this. My shipment from Prusa is taking a while to go out, so I'm not sure when the new sensor will come. I may try to test the old one to see if it works despite the physical damage.

TheZeroBeast, 2018-11-09:
Yes, interested to see if your sensor still picks anything up.

This issue is now resolved with Patch1 pegged for merge into master once other issues are resolved.

AbeFM, 2018-11-09:
Looking forward to trying it. I was hoping to have it today but now it looks like Monday at the soonest. Actually, Tue+ because Monday is a holiday.

closed 2018-11-09 by TheZeroBeast