TheZeroBeast/TZB-MMU2S-Firmware #7: Filament Trim Possibly causing "flakes"

AbeFM, 2018-11-06:
Master-Patch1
Version: 999-132

I've recently (last day or two) started getting weird tips, and plastic small round sections in the extrude's upper PTFE tube.

It's hectic around here so I'm not 100% on my profile, but it was weird, and these chips would weld to the tips so they'd end up really big. 
[image]

Anyway, with all that in mind, I noticed when doing a "load" that the snipped short of filament got left in the tube ahead of the trimmed filament (perhaps it pulls back slightly?) and anyway, then there's a loose piece generated, and they can be quiet short, a couple mm in the case I saw.

Maybe it's just normal bad tips and new to me, there's a lot of unknowns, but it coincides with reinstalling the blade.
So nothing to sweat yet but I'd like to note it in case others see it.

TheZeroBeast, 2018-11-07:
Hi AbeFM, this may have been something when I had cutting enabled which is no longer the case. I was getting cut pieces jammed in places as well and it was causing more problems than solving.

closed 2018-11-07 by TheZeroBeast

AbeFM, 2018-11-07:
Probably right to close it. I no longer think THIS is from grinding or chopping, anyway.