TheZeroBeast/TZB-MMU2S-Firmware #10: Suggestion: Always re-home idler & selector after every failure, or maybe even every filament change

jettoblack, 2018-11-09:
On your firmware, rehoming of the idler and selector is fast & quiet. Unless I'm missing something I see little to no downside to doing it more often, at least after every failure (as long as the FINDA reports clear), and maybe even on every filament change.

I noticed your README says this under failure modes:
* push middle button to rehome selector and continue (if idler suspected not synced press right to rehome and continue)

That seems confusing to me, how would I know to suspect the idler is not synced? And if I forget to press right, there's little chance of recovering from that. So why not always rehome? :)

TheZeroBeast, 2018-11-09:
Hey jettoblack,

From what I have noticed it isn't often that the idler gets out of sync and there is always a risk it won't rehome correctly.

At present there is a rehome at every 20 toolChanges of a print which is working good.

I have only had one time where the idler got out of sync on change 25 or so and extruded the wrong filament on load so I would have to manually feed to mk3-fsensor to continue.

If it is easier and is reliable that I can just have middle button rehome both idl and sel, we'll have to give it a go.

TheZeroBeast, 2018-11-15:
Hey AbeFM and Jettoblack,

Re-home after each fail is now the default on middle button with the latest release. MMU-2.1.3 & MK3-5.1.2.

Seams to be working good from my limited testing.

closed 2018-11-15 by TheZeroBeast