TheZeroBeast/TZB-MMU2S-Firmware #8: Filament runout fails as MK3 fsensor isn’t activated

TheZeroBeast, 2018-11-09:


AbeFM, 2018-11-09:
In you new version where you specifically activate the sensor? That seems backwards.

TheZeroBeast, 2018-11-09:
PRUSA comms layout is very basic and confusing, future would be full status updates to and from but that would take a lot of time.
At present the mmu2 says hey, I’m loading so turn on fsensor and let me know when you see filament.

I’m pretty sure the runout routine has mmu in a different state that disregards the turn sensor on request

TheZeroBeast, 2018-11-09:
Fixed by the following commits.
https://github.com/TheZeroBeast/MM-control-01/commit/8905f7e452cab6d52e4efc9ad9cf70f927f112db
https://github.com/TheZeroBeast/Prusa-Firmware/commit/c2c2fd3d2fa57b4232b1e99e005d3e877a7aca7e

closed 2018-11-09 by TheZeroBeast