TheZeroBeast/TZB-MMU2S-Firmware #13: Load Failure Timeout Pause Print

TheZeroBeast, 2018-11-10:
When a load fails printer remains ready for the MK3-FSensor to trip and pull filament in.

Implement a timeout to pause if in state longer than 5m.
Autorecover hotend heater, complete failed load once fixed and continue.

jettoblack, 2018-11-10:
That sounds pretty important. :)

TheZeroBeast, 2018-11-11:
Yes, good idea if I can. Not a code god by any measure.

jettoblack, 2018-11-11:
The Prusa FW already had this, if the MMU doesn't respond for a while, it will move the carriage to the back corner and shut off the hotend heater. Does it not work on yours because the MK3 is waiting for the filament sensor to trigger instead of waiting for a MMU response?

TheZeroBeast, 2018-11-11:
It does but when you’re not taking between each other, should be able to change it to work in another situation 👍🤔

TheZeroBeast, 2018-11-15:
This feature is now incorporated into the latest release MMU-2.1.3 & MK3 5.1.2.

5m timeout.

Clear MMU fault then clear MK3 fault, from testing this has recovered great so far 👍

closed 2018-11-15 by TheZeroBeast

AbeFM, 2018-11-15:
How do people feel about the 5 minutes? To me it was always too long.

TheZeroBeast, 2018-11-15:
This can be changed easy enough, it’s a MK3 default. Maybe 1m without communications.

jettoblack, 2018-11-15:
1 min is too short to notice and fix a problem. I'm ok with 5 mins since that's Prusa's default.

AbeFM, 2018-11-16:
My feeling is 3, but it is no hill to die on.

AbeFM, 2018-11-16:
Separate from that time length.. As I've now had a few failures with the firmware, often I find the MMU ready to go but nothing happens. I *think* it;s after I press the MK3's button, wait for it to reheat, and then it'll go? If so, I'd be happy to try to reword it from "press button on MMU" which I do a lot of times and nothing happens. :-)