TheZeroBeast/TZB-MMU2S-Firmware #14: Filament selector stays on wrong filament and won't change.

TimothyGold, 2018-11-11:
I ran into an issue where the print stopped and I found the selector on position 6 but the light was flashing on 3. The Mk3 display didn't say anything about being stopped. I pushed the middle button and pulled the filament from the 2 positions but the filament selector would not move. I tried to get it to rehome but to no avail. I then pushed the number 5 filament all the way to the extruder and the filament did print the number 3 layer with number 5 filament. In then pulled the filament back but the selector would not move. Number 3 light did turn green but selector stays on 5. I stopped the print and resetted which then started fine again. 
[image]
[image]
[image]

TimothyGold, 2018-11-11:
I will try again and see if I run into this again.  I think I did actually do what you stated in the clearing process.  The first filament I tried to pull was 5.  I don't know for sure if I saw the red/green clear before pushing the middle button though.  It never did rehome and never did move to position 3.  I am trying a new print that used all 5 colors this time.  One thing I noticed with your firmware is that it automatic rehomes when moving to position 1.  Does it ever rehome if filament 1 is never used in the print?  The print with the failure never did use position 1.  

Thanks again..

TheZeroBeast, 2018-11-11:
Hey mate,

Yep, keep an eye out for the flashing colours, if it doesn’t go green for a failed unload then there’s something keeping the ball bearing up.

Yes, both selector and idler rehome every 20th toolChange as well as selector rehome when it moves to pos 1.

TimothyGold, 2018-11-11:
I started a new 5 color print that almost finished without any problem until the last 1 or 2 layers.  This time I ran into another case where the selector ended up on the wrong filament number after a failure recovery... even after a rehome.  This time the flashing red light was on number 3 and the selector was on number 3 as well.  Also the MK3 was parked like it used to do in the old firmware.  I pulled 3 filament and then I think I pushed the right button.  It moved to do a rehome on the right side and then moved to the new filament.  The only problem is that the light moved to 4 but the selector moved to 5.  It loaded the filament down to the extruder from 5.  There is another issue where after it reached the extruder MK3 did not know to start going again.  I had to push the middle button and after the hot end heated up again I had to manually make sure the filament fed in.  So it finished this step with the wrong color.  I continued to move on and I think because the next color was 1 it was able to do another rehome and then finish the print correctly.  I got real close to finishing a print without issue!  I think this might be related to the problem listed by  @jettoblack.  So in my opinion a few concerns.

1.  Seams that homing does not always home correctly.
2.  There is some disconnect between the old way MK3 parked and knowing to start again.
3.  I wonder if there is a print where filament 1 is never used and homing only happens every 20 cycles this may not be enough to save a print.  I think in this failure it was the fact that filament 1 was the next one which forced a rehome.  

Hopefully this helps.  Thanks again for your work.

TheZeroBeast, 2018-11-12:
Ah, that's a shame.
Interesting that your selector got out of sync, it has always homed correctly on my setup unless there was filament in the path.

I am trialing some tweaking for the idler drum that has proven to make perfect at homing from what I have seen. Still printing my trail print with it. ~8h in.

Yes, there may well be, there are three states MMU can fail in so I may need to drop that count so MK3 recovers after pause more reliably.

I haven't ever seen the selector get out of sync but to change it to 10 or something else is very easy if you'd like to test that setup.

TimothyGold, 2018-11-12:
Maybe I have a unique build.  The selector does move smoothly and I don't know what might be causing this either.  I think reducing this number might help especially in case this does happen again and the filament 1 is not being used.

AbeFM, 2018-11-13:
Two things I've noticed:
1) wiggling the cables on the back of the unit changes how likely it is to
reset/initialize correctly.
2) the drum is less likely to home correctly if there is filament under the
rollers (not quite into its tube)

I think the filament pulls back too far (maybe due to string or dirty ball
sensor), then drum gets filament cocked sideways and stalls during/after
homing.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2018-11-15:
Hi TimothyGold,

Please try the latest release, there are improvements to homing and the way it recovers.

Let me know if you're still having difficulties.

Rob

TheZeroBeast, 2018-11-18:
This issue no longer seams to be an issue, if you're still having this issue on the latest release please reopen with further details so I can assist further.

closed 2018-11-18 by TheZeroBeast