TheZeroBeast/TZB-MMU2S-Firmware #15: Idler re-homes to wrong position when MMU closed, works correctly when open

jettoblack, 2018-11-11:
Trying 201 / 2132 and just experienced a new problem. I turned the printer on with the MMU body closed, and the idler and selector both homed correctly. The print started fine, and the first 20 or so filament changes were successful. Then when the printer tried to switch from 4 > 3, I looked over and saw that the selector was on 3 but filament 5 had been driven out the front about 10cm, and the MMU light 3 was blinking (failure to load filament 3, as the FINDA was not triggered). The idler was on the position to drive filament 5, when it should have been on 3.

I opened the body to manually retract filament 5, then closed the body and pushed the right button to do a re-home and reload. The idler did the re-homing motion, but ended up in the wrong position. When trying to load filament 3 it was moving to some in-between position that didn't drive any filaments.

What I had to do is open the body, press the right button to re-home the idler with the body open, then I could see it correctly moved to the filament 3 drive position. So I closed up the body and hit the middle button and was able to resume the print. The next several filament changes have gone fine. I'll update if anything changes later. :)

Any idea why the idler would home correctly on initial boot, but during the print go to the wrong position and then fail to re-home correctly with the MMU closed, but did re-home correctly when opened up?

TheZeroBeast, 2018-11-11:
Hey Jetto,

How far are you screwing the idler bolts?
I have been doing between 1.5 - 2 full turns past flush with top surface or else homing doesn’t find home due to too much resistance on the bearing lumps.

That’s the exact reason I put the rehome on right button. I went twenty changes manually before it homed correect and finished that print.

jettoblack, 2018-11-11:
My tension screws are just barely past flush, maybe 1 turn past flush at the most. I checked to make sure there weren't any filament bits or something else causing the idler to stick but I didn't see anything, and I checked the two screws that hold the idler to the motor shaft are as tight as comfortably possible. The idler motor is running just barely warm.

TheZeroBeast, 2018-11-11:
Hey man,

I just did some tuning and now have accurate homing even with screws in 4 turns, can you try it out please? I'll make a custom branch

TheZeroBeast, 2018-11-12:
ah ok man, interesting. We must have something different. Idler touching the case or something maybe.

I'll let you know which parameters I tried changing, you might be better off having a crack and letting me know how you went.

In config.h the parameters to change would be the stallguard value for idl, I think fro 213 it is at 6
and the homing currents play a roll as well. I dropped the IDL(third) current to 28 as I found with my setup with the normal running current of 35 stallguard wouldn't trigger and it would just loose steps(make noise) at home.

My process with mine was to tighten bolts 4 turns past flush then adjust value, flash. If in a continuous rehome loop then it's tripping stallguard(sg) and rehoming so increase the idler sg value.

If then, like you have to stepper isn't triggering sg before it actually stalls then the sg value needs to be decreased.

These values are sensitive to stepper temp, I haven't had mine at room temp to try from that state but that may also be the difference.

jettoblack, 2018-11-13:
@TheZeroBeast I went back to 212 and successfully printed a 5 hour, 3 color print with 90 tool changes and it went perfectly. Now I'm trying a 24 hour print with 700 tool changes. :) This issue can probably be closed as user error but I'll update you tomorrow how it goes.

AbeFM, 2018-11-13:
I just got the printer sorted in master patch 1 & Prusa printer fw.
Optical sensor showed up.

Just pull master of both? Should I wait for next release?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

jettoblack, 2018-11-15:
I'm closing my issue as it was a physical problem with the idler and not caused by FW.

closed 2018-11-15 by jettoblack

AbeFM, 2018-11-15:
Not mine it should be a priority, but since the screws DO come loose I
driver if you could test for it - full left then full right or similar.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes