TheZeroBeast/TZB-MMU2S-Firmware #17: First layer cal not working

jettoblack, 2018-11-12:
Here's what happens when I try to do a First layer calibration:

The printer starts warming up, and asks which filament to use. I select Filament 1.

The MMU2 loads filament 1 down to the sensor, then starts blinking (load failure).  I guess it did not receive the expected sensor response from the MK3? The printer stays at temp but doesn't do anything.

Once in this state, I can't figure out how to get out. The MK3 is in "Print mode" (LCD menus are as if there is currently a print running), but there is no cancel print option, nor can I eject/unload filament in this mode. I had to open up the MMU to manually retract the filament, then do a hard reset on the printer.

Unfortunately my current Live Z value is not good after adjusting the PINDA, so it looks like I will have to flash Prusa FW to the MK3/MMU to get past this.

TheZeroBeast, 2018-11-12:
Yes, first layer calibration wizard has not been touched and will not work. If someone has time to retrofit this code they can. At this stage, have printer setup on stock and don't earse eeprom on printer to reset the settings.

mylife4aiurr, 2018-11-12:
I wish I knew that before I installed your firmware

mylife4aiurr, 2018-11-12:
Ok no problem. Just note it somewhere, so users knows. It's a deviation from the standard firmware installation process. By resetting to factory I erased my 1st layer offset. At least have user take note of old zoffset value.

######
Prusa is about to drop mk3 3.5rc firmware. Looking at the commits it has some new mmu2 code???

TheZeroBeast, 2018-11-12:
Yeah, will do for sure. Good pick up too, thank you.

Yes they have added a command for MMU2 but it doesn't relate to using the MK3-FSensor or fixing any known MMU2 issues that I can see.

They have added a menu to enable/disable autodeplete in the tune menu when printing which I plan to pull across when I have some time.

jettoblack, 2018-11-12:
Here's a workaround so you don't have to re-flash: Take the bowden tube off at both ends, stick some filament into the FINDA, then power off & on. The MMU2 starts up in panic mode because the FINDA is triggered, so the MK3 won't see the MMU2 installed. You can use it just like non-MMU mode.

mylife4aiurr, 2018-11-13:
Since it's relevant.... to my earlier question.
Prusa fixed/made some changes to first layer calibration in release candidate mk3 firmware 3.5.0rc specifically for MMU2.

TheZeroBeast, 2018-11-15:
Hey Mylife4aiurr,

I have just merged all 3.5.0rc1 code into my repo but haven't tested first layer cal, I see they have a section within the wizard that requests filament from MMU but might need a small tweak to get it working with the FSensor.

mylife4aiurr, 2018-11-15:
Excellent news. Factory reset between firmware updates really is the best path. Just 4 consistently I'd do 1st layer Cal, but at least now I know to load stock then yours no reset.

jettoblack, 2018-11-15:
In 5.0.2 / 2.1.3 FW, after starting First layer cal and selecting a filament, it loads the correct filament down to the sensor, but it appears the sensor doesn't trigger or it doesn't get communicated to the MMU2. The MMU2 starts flashing red load error and none of the MK3 LCD options or the MMU2 buttons will do anything at that point.

mylife4aiurr, 2018-11-15:
@jettoblack 
I dont think zerobeast tweaked yet for work with FSenor

TheZeroBeast, 2018-11-15:
Hey Jettoblack,

Thanks a bunch for testing, I’ll find what the heck is happening today and let you know👍

Wow AbeFM, trying tpu, sweet.
Will have to use the filament types soon as they’re already passed to mmu just not used. Did you slow loads down at all for your tpu?

AbeFM, 2018-11-16:
I've gotten prints with TPU and pet wiring with the same settings.

Having issues (check forums), but don't seem to follow material, only
seeing issues in PLA. Also seen some grinding, is distance to bondtexh
wrong - or timing?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2018-11-16:
Ah, that is so cool. I haven't tried my 87A TPU again since I tried with stock MMU2 code. I'll try give it a go again this weekend.

We still need a matrix of speeds for the main filament types, even if only small adjustments.

I also found what was stopping First Layer Cal and will test as soon as I can between prints.
FSensor function is normally is disabled for all Wizards, I have enabled it for the period of the load required for MMU.

TheZeroBeast, 2018-11-18:
I can confirm first layer cal now tested and working, it'll be in the next version shortly.

closed 2018-11-18 by TheZeroBeast