TheZeroBeast/TZB-MMU2S-Firmware #18: After initial homing, selector stalls

Panayiotis-git, 2018-11-12:
In motion.cpp, at the method homeSelectorSmooth(), the final movement of the selector to the 1st filament is done with very high speed and the selector stalls.

return moveSmooth(AX_SEL, SELECTOR_STEPS_AFTER_HOMING, 8000, false);

speed = 8000.
It should be le 3000.

TheZeroBeast, 2018-11-14:
This is not a problem for a lot of people. You can manually change that value and the MAX_SPEED_SEL if you find them too fast. Or see if your blade is catching and remove it.

closed 2018-11-14 by TheZeroBeast

Panayiotis-git, 2018-11-14:
You are right, I can manually change the value and this is what I did.
It bothers me though, that even if we define the MAX_SPEED_SEL value, the homeSelectorSmooth() do not respect this limit, but instead it uses a value much larger that the limit.