TheZeroBeast/TZB-MMU2S-Firmware #20: Pulley grinding filament during loads

jettoblack, 2018-11-15:
I'm using the Slic3r settings from the readme, but I still get filament grinding on the MMU pulley during each load, which leads to lots of filament shavings building up in the MMU. I get some loud clicks/thumbs as the pulley is still pushing after the filament is already in the Bondtech gears. Seems like maybe the number of steps required to load between the optical sensor to the Bondtech gears is lower on mine than on yours? Might be different diameters of the hobbed gears. I can try tweaking the number of steps in the FW, but a more robust solution might require some kind of calibration procedure.

TheZeroBeast, 2018-11-15:
Hi jettoblack,

Yes, I noticed this also and have found the culprit, I just did a release that should improve if not eliminate this. MK3 was not waiting for the completion of the steps to bond tech before starting to pull into MK3.

closed 2018-11-15 by TheZeroBeast

jettoblack, 2018-11-15:
This seems to work so far, although the delay could probably be made a bit shorter.