TheZeroBeast/TZB-MMU2S-Firmware #22: Logs of filament unload failures.

AbeFM, 2018-11-16:
I';ve replaced "echo:busy: processing" with a "." to keep the logs useful.

There are a couple missed events in there. I believe they were feeds that went wrong here.
[Log1 two resets at end.txt](https://github.com/TheZeroBeast/MM-control-01/files/2587768/Log1.two.resets.at.end.txt)
At the very end of this, I reset the MMU with its button, and you see the coms overflow.

The next one is much the same, it's misfeed is:
`NORMAL MODE: Percent done: 2; print time remaining in mins: 269
SILENT MODE: Percent done: 3; print time remaining in mins: 269
NORMAL MODE: Percent done: 3; print time remaining in mins: 268
SILENT MODE: Percent done: 3; print time remaining in mins: 269..............
echo:Advance K=60.00
 E/D=Auto
echo:Advance K=0.00
 E/D=Auto..
MMU => 'CMD ACK 0x11'
MMU <= 'T1'....
MMU => 'waiting for filament @ MK3 Sensor'........................................................................................................................................
MMU not responding.......................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
......................`

I would find the filament well into the heatbreak (not problematically, but thats where it is).
 Wish I had more for you. It's working for a hand full of changes without issue. 

Unsure yet if I'll keep running it without something specific to do/look for, Maybe my tips just needed forming, :-)

TheZeroBeast, 2018-11-16:
Hey AbeFM,

This is definitely the MK3 not advising the MMU of filament being sensed.
The MMU doesn't respond within 5m then as it hasn't been asked to.

Is this still with the broken glass FSensor man or do you have the replacement in?

I am still having to blow dust/particles off the glass every 1000 changes or so.

AbeFM, 2018-11-16:
brand new sensor. It ten-ish times on two PLA'S, TPU and pet. I had issues
with PLA (weirdbut it's what happened). I wonder if it is same chip as
before. The company only lists 850 nm devices.

Back on MP1. Which means back to hearing the machine grind for however long
it takes you to stop it, at full temp. I doubt you want to maintain two
forks but a "three tries and stop till button press" might be good.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2018-11-18:
Hey AbeFM,

Ah nice, new sensor, that was quick.

I had a couple of random issues with the FSensor locking up which I think what you're seeing above might be.

I am testing some ways of catching this and reseting and re-initialising the FSensor.

Yes, not ideal, and I don't have much time to go find MP1 and start modding it.
Hopefully I'll get a couple tests done between work this week and have a solution to the random FSensor lockups.

AbeFM, 2018-11-19:
I'm having an MP1 issue now where it'll unload on Fil0, and keep rehoming the drum over and over till reset.
Maybe it's due to newer (OEM RC) pinter FW. Not sure what triggers it, but I suspect it may be an interaction between MP1 and that new FW.

I'm out of town till next weekend, looking forward to trying new stuff when I return.

TheZeroBeast, 2018-11-21:
Ah ok, yes, there's possibly issues between MP1 and the new PrusaRC.

Give the new code a try, I think I have almost squashed the unload bug where it appears nothing is wrong.

TheZeroBeast, 2018-11-30:
Resolved in next release 2.1.5.

Clean sensor testing and LCD notification will also be included based on premature triggering directly after enabling.

closed 2018-11-30 by TheZeroBeast