TheZeroBeast/TZB-MMU2S-Firmware #23: Problem

Dye007, 2018-11-20:
Hello,I have uploaded the firmware now, unfortunately this does not work at all. Always problems with the filament change.

https://www.youtube.com/watch?v=U5_4eYT-Ie4&feature=youtu.be&fbclid=IwAR1n0nHQbJB56YHFY7e5nFVb2C2MMaHPapPdPfro3QNQqK147IoP6AwWmwg

https://www.youtube.com/watch?v=YwjcyqjXsfY&feature=youtu.be

TheZeroBeast, 2018-11-21:
Hi Dye007,

Please try the latest release, read through the README.md with some points to take note of.

If you're still having issues please get serial output of printer when issue occurs and a description of what state the issue occurs. e.g. first homing after power. Loading filament, unloading. etc.

Allram, 2018-11-22:
@Dye007 have you removed the blade? I had the similiar problem as you until i removed the blade. Was perfect after that :)

TheZeroBeast, 2018-11-23:
No response from originator. If you're still experiencing an issue after following provided advise please advise.

closed 2018-11-23 by TheZeroBeast