TheZeroBeast/TZB-MMU2S-Firmware #25: Resume on failed load

madxbgt, 2018-11-22:


TheZeroBeast, 2018-11-22:
Please provide extensive detail of your issue, photo/video so I am able to track down the issue.

madxbgt, 2018-11-23:
After printing several layers the MMU re-homes itself and then waits at the correct position with a the green led flashing and will only resume when the button pressed and most times will resume again.
However when it moves to the next color the bowden selector stalls trying to engage the filament, I have loosened the clamp screws to the point where it is almost disengaged with the nut and the selector will stall, if it does manage to engage after several resets (or crash) it will continue for several color changes again only to fail again after it re-homes.

TheZeroBeast, 2018-11-23:
Thank you for the description.
I’ve had a read and watched the video 👌
It appears idler(drum) and selector homing is working fine. The selector is definitely having issues running at that speed.
2 ways to fix this for your situation.
First is you could just change the max_speed_selector down to say 5000 from 8000, flash and try.
Second would be to try optimise the selector. Remove blade as it randomly catches on the case if not installed just right. With the printer off gently move selector left and right and feel for places it catches, might even just need a drop of oil on the two rods and screw.

madxbgt, 2018-11-24:
Thanks mate,
I was thinking the same have removed blade and oiled slides and selector moves nicely, will try dropping the max speed.

TheZeroBeast, 2018-11-25:
No cmakelist or anything like that needed. You can upload direct through Arduino IDE over usb to the MMU2 board.

closed 2018-11-25 by TheZeroBeast