TheZeroBeast/TZB-MMU2S-Firmware #26: Feature request: Gcode to disable/enable steppers on MMU2

jettoblack, 2018-11-22:
The MMU2 steppers are fairly noisy when idle, this is on a MK3 and I understand it's even worse on a MK2.5 due to using 12v. They are kept idle to avoid the need to re-home when turning them back on, which on the Prusa FW was a good idea because re-homing was slow and not totally reliable. On your FW re-homing is now relatively quick and reliable (it better be since you're rehoming after every N tool changes), so there's no reason to avoid re-homing. Still, I think disabling the steppers needs to be prompted by a Gcode command and not by something like an idle timeout. 

For single material prints you could put the "turn MMU2 steppers off" command after the T? command to quiet things down during long single material prints, and then turn them back on at the end before the final unload.

In the case of multi material prints you probably don't want the steppers turning off during the print, but you could at least turn them off at the end.

TheZeroBeast, 2018-11-23:
hey man, good idea. I'll see what I can do.

AbeFM, 2018-11-24:
One would imagine doing it in the slicer, where you know how long it will
be off for, would make sense. But as  you say, the cost is low here and
it'll take less time to fix than to get the suggestion acknowledged. :-)

TheZeroBeast, 2018-11-30:
Hey AbeFM,

15m timeout when not loaded and off @ final unload of print now disables all steppers.

Pegged for release in 2.1.5.

Required rewrite of shr16 methods and defs as they were inacurrate in stock.

TheZeroBeast, 2018-12-08:
All tested and working in latest 2.1.5 commit by means of disable on unload e.g. end of print and or if unloaded for more than 15m e.g. after you turn on and don't print straight away.

closed 2018-12-08 by TheZeroBeast