TheZeroBeast/TZB-MMU2S-Firmware #28: Laser sensor destroyed by Black filament not being sensed

TimothyGold, 2018-11-24:
I tried the latest release and was having good success until on a black filament load the mmu2 kept pushing the black until it coiled up in the extruder. After cleaning it out I found the filament destroyed the sensor. I think ultimately this is why prusa will never be able to reliably use this sensor in its release. I appreciate all the work you have done and hopefully some of your other features make it into the prusa build. Thanks again.

TheZeroBeast, 2018-11-25:
Thank you for your sacrifice. Hopefully prusa will eventually give us all the laser sensor promised all MK3 owners.

closed 2018-11-25 by TheZeroBeast

AbeFM, 2018-11-25:
My first sensor broke before I tried this firmware. I bought a replacement
so I could try it.

I put it in, and it looked fine. A few days' use later, my new sensor is
broken. I don't think anything hit it, so I'm a little confused.