TheZeroBeast/TZB-MMU2S-Firmware #29: Odd behaviors, loading.

AbeFM, 2018-11-25:
Once, after I reboot everything, press right button once on MMU, then center button, instead of loading filament like on the other channels, it rehomes. The other channels worked as expected. I've been unable to reproduce.

It seems like loading is failing, I'm using a filament I thought worked before - though it's a lighter TPU. It wouldn't trigger the sensor, I'm not sure it's coming out far enough.

If I unscrew the bowden and put in PLA, I can get it to trigger if I do it at the right time.

It would be helpful if seeing or not seeing stuff on the sensor showed up in the serial logs - there's a lot of overhead in just trying a filament/movement/speed/angle, whatever, it's 10+ minutes to start a print, and since the sensor stops getting watched you can't just hit load and do your testing.

So far MP1 seems like th best - there's a lot of great new features but it's a pain. Now I have to flash back to get the tube length's tweaked, flash back again, just to see if maybe that helps the sensor trigger. Actual sensor feedback would be better if you can think of how to give it.

TheZeroBeast, 2018-11-30:
Setting length in a previous compile shouldn't change the hardcoded length in rel 2.1.4

2.1.5 feature I'm working on is a set end of bowden tube for PLA for those that have different than stock length.

> It would be helpful if seeing or not seeing stuff on the sensor showed up in the serial logs - there's a lot of overhead in just trying a filament/movement/speed/angle, whatever, it's 10+ minutes to [...]

Are you saying you'd like to be able to pas filament into the top of extruder and see if it triggers when MK3 isn't printing??

This is an interested concept, I would say it would be possible to have the sensor always checking when not printing as per normal MK3 although just display a message for a couple seconds on a trigger instead of a load.

What are you're thoughts on it?

AbeFM, 2018-12-05:
Yep, that's what I was saying. There's a "pinda status" in the menus I haven't checked yet, perhaps it would work.

If for no other reason than I'm tired of buying sensors, it's worth having an intermediary. In addition, it works better. I've been running one for a bit now with your latest code and it's working well. A way to set the distance without compiling would be cool but isn't really needed just now. :-)

TheZeroBeast, 2018-12-08:
All tested and working in latest 2.1.5 commit.

When not printing you can now detach Bowden tube from extruder end and test any filament, a message is shown on display when sensed.

closed 2018-12-08 by TheZeroBeast