TheZeroBeast/TZB-MMU2S-Firmware #31: Loading Issue

megabytemike, 2018-11-26:
I'm having an issue after moving from the standard Prusa firmware to the latest version of TheZeroBeast. When loading a filament to nozzle or when it loads itself via a gcode, it selects the correct filament then loads it perfectly all the way down to the extruder but then it stops and starts flashing red on the currently selected filament and errors out, it clears when i manually remove the filament from the extruder back past the selector (finda) then it loads it again and same error. I can not get it to get past this error, not sure if I did something wrong when flashing this firmware. I have the extruder Fil sensor on and have changed my Slic3r settings to match the readme. (MK3)

megabytemike, 2018-11-26:
update, checked extruder gears during load and they never turn to grab the filament once the filament reaches the bondtech.

TheZeroBeast, 2018-11-30:
Hi Megabytemike,

Thanks for posting your issue. I haven't seen this one before.

Just to confirm, you have flashed my corresponding MK3 firmware as well as the MMU2 firmware?

Evildr14, 2018-12-01:
I to am having the same issue. The filament sensor in the MK3 is not picking up the filament. 

23:19:10.323: tmc2130_goto_step 0 29 2 1000
23:19:10.412: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x01
23:19:10.532: tmc2130_home_enter(axes_mask=0x02)
23:19:15.270: 0 step=33 mscnt= 541
23:19:15.274: tmc2130_goto_step 1 33 2 1000
23:19:15.283: tmc2130_home_exit tmc2130_sg_homing_axes_mask=0x02
23:19:45.037: Unknown G code: $296 G21
23:19:45.039: MMU <= 'F0 0'
23:19:45.072: MMU <= 'F1 0'
23:19:45.207: MMU <= 'F2 0'
23:19:45.340: MMU <= 'F3 0'
23:19:45.443: MMU <= 'F4 2'
23:19:45.720: MMU => 'CMD ACK 0x11'
23:19:45.724: MMU <= 'T1'
23:19:46.281: MMU => 'waiting for filament @ MK3 Sensor'
23:21:25.553: MMU => 'CMD RETRY 0x81' (85)
23:21:25.973: INT4

I have will swap the sensor out of my second MK# once it finishes printing to see if that makes any difference.
[image]

AbeFM, 2018-12-01:
That was my experience. I attributed it to broken sensor, but it worked in
other cases.

Evildr14, 2018-12-01:
I changed out the filament sensor and i can get orange and white to work but all others still fail. Not to sure what i can do now....
[image]

megabytemike, 2018-12-01:
Yes, I flashed both several times and went back to stock to verify my sensor was working. I even tried the earlier release with the same problem. Wonder if it has something to do with the different sensor revisions.

Evildr14, 2018-12-01:
I just started printing the indirect sensor mod from martin_au on thingiverse (https://www.thingiverse.com/thing:3223513). I'm not sure if it will change the filament sensing distance but I'm keen to give it a go.

TheZeroBeast, 2018-12-03:
That bearing sensor mod is a great idea, keen to do it when I find time. I hear PRUSA has a new sensor they're working on, hopefully we MK3 owners will get it complementary or even first go at them if it gets a solid filament sensor. PAT9130EW I do believe.

Evildr14, 2018-12-03:
Well I'm almost set. I have printed out the parts for the bearing mod just waiting on the bearings. With the inconsistencies of the current setup I dare say they will be putting a bit of effort into it. Unfortunately  with the number of MK3s in the world now, anything they release would take months to arrive.

AbeFM, 2018-12-03:
I found the bearing to be loose in the housing, so I'm trying to narrow it up, and redo the top to get a better bowden tube set up.

Evildr14, 2018-12-03:
@AbeFM Nice. let me know how you go.

AbeFM, 2018-12-03:
I'm using this one: https://www.thingiverse.com/thing:3129921 - it will definately need the distance from sensor to pullys changesd, so either way there is a firmware change, I guess.
It keeps me from having to reprint the housing or tweak the firmware. I have to admit, once I get into Git more than "commit" or "pull" it stops makes sense and starts taking hours. I've been trying to get the commits you mentioned but it has become a lost cause.

Evildr14, 2018-12-07:
My understanding of git is rather limited at this stage. I tried to build the indirect filament sensor of thingiverse without much success. If the bearing is too close the bearing will turn but won't let the tip through and if it is too far away the bearing will slip on the filament. I started designing my own which seems to work quite well but adds a little weight. I figure if I'm going to change the firmware I might as well add a BMG extruder.

Evildr14, 2018-12-07:
[image]
[image]

TheZeroBeast, 2018-12-09:
How goes your new sensor setup Evildr14?

Evildr14, 2018-12-11:
Slow and steady progress, my final bits arrived in the mail today...the new pancake motor. I almost had tears drilling out the new BMG to allow filament tips through....the tolerance of the bmg is way too tight.

Evildr14, 2018-12-11:
[image]
[image]
[image]

TheZeroBeast, 2018-12-11:
Looking good none the less but yes, it would of been hard drilling something so nice and new.

Also, this issue thread is now resolved with the latest RC code. I will rename so you can share your progress with the BMG.

AbeFM, 2018-12-11:
To set Optical Filament Sensor to Bondetch Distance, I modified line 32 of Motion.cpp:
const int STEPS_MK3FSensor_To_Bondtech = 756; //Abe 385 original number, I need to modify 385 by 191% (32mm/16.5mm) = 755.xxx

Just to tie up the thought.

Looks like an interesting build.

TheZeroBeast, 2018-12-15:
Thanks for the update, keen for the test print results 👍

AbeFM, 2018-12-17:
Did you try a grooved pulley? I'm pretty convinced but haven't built one yet.

AbeFM, 2018-12-19:
Get something generic on Amazon? The last time I ordered one, I ordered 2 just in case. And put a connector on for easier changes. :-)

Evildr14, 2018-12-20:
I will definitely order a couple this time round. https://www.thingiverse.com/thing:3299790 is the files if anyone is keen for a look.

Mjankor, 2018-12-31:
I had this bug. To resolve it I disconnected the MMU and checked that the printer was working properly under normal operation (so checked autoload and filament sensor status). I also disabled and re-enabled the filament sensor. I then plugged the MMU back in and reflashed it. This then worked, with "triggered" showing up on feeding filament into the top of the extruder. I think that when the MMU is connected, the filament sensor is still disabled, even if it says it's on in the menu.

TheZeroBeast, 2019-08-20:
stale.

closed 2019-08-20 by TheZeroBeast