TheZeroBeast/TZB-MMU2S-Firmware #33: Filament runout can cause issues

danielegger1, 2018-11-27:
I am not sure if I should also report such issues to you, as Prusa is doing nothing at MMU firmware I hope you will correct issues ;).
1. issue:
So I had a filament runout and the filament was moved back but I was not able to get the filament out of the tube. In this case the filament needs to move forward to get it out.
2. issue:
I loaded the new filament into the mmu but adter that the mk3 firmware got stuck and I wasn‘t able to continue. This bug is a little bit hard to describe but with several tests this can be easily checked from a programer ;)

TheZeroBeast, 2018-11-30:
Please specify on which version of my code you're having issues

danielegger1, 2018-11-30:
Surely on the latest ;)

TheZeroBeast, 2018-12-08:
Fixed eject methods and tested and working in latest 2.1.5 commit.

closed 2018-12-08 by TheZeroBeast