TheZeroBeast/TZB-MMU2S-Firmware #38: Calibrate distance from sensor to gears during bowden length cal.

AbeFM, 2018-12-11:
You're already doing a bowden length cal, and I assume you see the sensor trigger during this.

It sounds pretty trivial to grab that number as well out of the process. Let me know if you need psuedo code, but I think it speaks for itself.

AbeFM, 2018-12-11:
Looking at the new code, either you have to do this, or perhaps set just a multplicative factor for those differences so you don't have to do SO MUCH.

Maybe you could go as far as to use the k-factor to set this distance, but to start, having one number that describes the difference which the user sets (distance from sensor to gears) and then a factor for feeding (even if different from main feed and sensor-to-gear feed) would be less numbers for the user to do.

`int8_t filament_type[EXTRUDERS] = { 0, 0, 0, 0, 0};
const int filament_lookup_table[8][3] =
    /*0*/ {{4600,   400, 2800},
    /*1*/  {4000,   100, 1500},
    /*2*/  { 300,   330,  300},
    /*3*/  {-610,  -610, -610},
    /*4*/  {6000, 10000, 6500},
    /*5*/  { 600,   300,  550},
    /*6*/  { 350,   200,  385},
    /*7*/  { 400,   200,  455}
};
/**
 * [X] == variables based on type
 * [Y] == filament types (0: default; 1:flex; 2: PVA)
 *
 * [X]
 *  0   MAX_SPPED_PUL                  S/S
 *  1   ACC_FEED_PUL                   S/S/S
 *  2   STEPS_MK3FSensor_To_Bondtech   STEPS
 *  3   FILAMENT_PARKING_STEPS         STEPS
 *  4   FSensor TIMEOUT                MS
 *  5   FEED_SPEED_PUL                 S/S
 *  6   L2ExStageOne                   S/S
 *  7   L2ExStageTwo                   S/S
 *
 */
`

Right now it looks like I'd have to set 300/330/300 to 755/830/755. Cleaner would be "22 mm", which then gets multiplied by steps/mm, and possibly modified for k-factor. Same for the speeds, high k-factor materials could be fed/retracted more slowly.

TheZeroBeast, 2018-12-11:
the Bowden calibration is not using sensor it is a manual eye process with Bowden removed from extruder.

it isn't a good idea to use a step/mm as we know there is a wipe variance in the extruder gears.

All these variables I feel need to be hard as some directly correlate to the Slic3r speeds and can't just be changed without creating issues.

It also isn't a space concern as I'm not using any where near as much flash as the previous FW/STOCK FW used due to the serial comms re-write.

AbeFM, 2018-12-11:
I got the newer gears, so it stopped being an issue for me. A bit selfish. However, if that's the case, why aren't you doing a cal on each gear in the calibration routine?

What I was trying to say was: 
1) Start a manual calibration
2) Printer shoves out ~350 mm of filament
3) At ~325 mm, the sensor is triggered. Record that value
4) User sets the total length

I propose to follow up by setting the sensor-to-gear distance by subtracting the steps to sensor triggering from the total stepped distance to gear.

Similarly, a menu to enter the number or a g-code to set it, but I figure the above would be better?

AbeFM, 2018-12-12:
I'm selfish because I stopped caring about different gear sizes when mine stopped being different sizes.

> I don’t need to as fsensor is used over a 6s window, this covers discrepancies
I hate to say it, but if you're using the calibration as a safety, then you should think about it? It's more than 1.5 cm difference on feed, if you're going blind. Again, I'm ok with it, now, but with the 1st set of gears it would be a (corner case) issue.

TheZeroBeast, 2019-01-13:
This has been adding in 2.1.6, feel free to give it a go.

It is within the **SETUP** menu as a part of the Bowden Length Calibration.

https://github.com/TheZeroBeast/MM-control-01/releases

closed 2019-01-13 by TheZeroBeast