TheZeroBeast/TZB-MMU2S-Firmware #42: Post loading Extruder Clicking

AbeFM, 2018-12-11:
First off: Awesome, it recovers after a failure. Filament didn't load, since there's no cutting, it gave up nicely, sitting to the side to allow me to clear filament, and then it rehomed and went to the right spot.
So... Sensible. :-)

On first load (but also on first material change (likely more to come)) the printer ejects too much filament!

Could it be the "-13" on "extra loading distance".

TheZeroBeast, 2018-12-11:
Hi AbeFM,

👍👏 nice

For the too much you say, not sure why it is different to mine as I would say the extrusion starts at a good time which on a toolChange I consider to be just before the faster purge movements, I notice a pop as the are pocket is expelled with new filament.

You can increase the deficit with that Slic3r setting to not load as much after mmu2 load is complete.

AbeFM, 2018-12-11:
I've seen this on and off since I got the MMU. It's something in how it handles the purge tower. You get a blob, which causes a crash every few purges. Eventually you melt down the blob but there's just no reason to have it.

It's the very first issue I posted about the MMU and you can imagine how much of a shit Prusa gave.

You're saying turn the 0 back to a -13 for the "extra loading distance" on the Printer Settings?

I can give it a try but it seems stupid to me to have it programmed to ruin the print then use that variable to re-zero. Further, I didn't notice a change when I altered it from your (and Prusa's) -13 to 0. Maybe I'm just not testing closely enough, but if that's really the answer I feel like I'd have to run ~-25 to get it right.

Your thoughts?

TheZeroBeast, 2018-12-12:
Yes, increase that value to decrease the amount of load... plastic

If you watch from the size of the extruder the filament should just arrive at bondtech when bondtech themselves start turning.
If more is pushed in first it will be losing more.

This would be the distance after FSensor value in the ftype matrix.

TheZeroBeast, 2018-12-13:
Related to incorrect Bowden length cal point.

closed 2018-12-13 by TheZeroBeast

AbeFM, 2018-12-13:
I think the bulk of the problem was my not following the calibration procedure.

I have noticed extra filament only on channel 2 (the center one), with excess material on wipe tower and grinding, but have yet to investigate it enough to have anything to say about it yet.

We'll see if I make any progress today.