TheZeroBeast/TZB-MMU2S-Firmware #43: Loading doesn't work

AbeFM, 2018-12-11:
Every time the printer tries to do a filament change, the filament is unloaded, then the new filament is not fed.

If I don't have all the filaments pre-loaded, the rehoming fails and the drum ends up near/at the homing location.
If I DO have them loaded, it homes correctly and goes to correct filament, but then doesn't see it at the sensor.

Not sure how to help with this - perhaps serial logs?

AbeFM, 2018-12-11:
I suspect the printer isn't responding to the MMU with it's "filament received" status when in error mode. Not based on anything, just a whisper in my ghost.

TheZeroBeast, 2018-12-11:
Hmm, I haven't had this where the sensor isn't responding.

If you could get the serial log and the toolChange number it happened on I can take a deeper look.

TheZeroBeast, 2018-12-11:
👍 sounds good.

TheZeroBeast, 2018-12-12:
Duplicate to #44

closed 2018-12-12 by TheZeroBeast