TheZeroBeast/TZB-MMU2S-Firmware #45: Bring Back the Blade!

AbeFM, 2018-12-11:
I know there's some crow for me to eat on this one, but really, in running MP-1 for a month, I'm sold on the filament cutting. It works really, really well. All sorts of stuff that doesn't feed the first time feeds just great after a cutting.

I'm not sure how it fits into your world-view of MMU operation, but the routine you had never steered me wrong.

TheZeroBeast, 2018-12-11:
In my experience I never got a load with a sharp edge, it would catch just after fsensor before bondtech gears.

Feel free to fork and add it back in for a personal build. 👍

AbeFM, 2018-12-12:
Oh yeah, things are a mess in there. I'd go with one of the indirect sensor mods.

Other than smashing two sensors (!), I really found the cutting to work well. I originally had the same concerns, that the blunt tip wasn't super useful, but over the past couple of weeks I've had it work many hundred times without issue, and I don't think any failures.

The other issue I get is filament creeping too far out and catching the selector at random times.

It does seem it would hold back your testing to put it back in. Is it as simple as commenting a line of code? Can  you point me to where?

TheZeroBeast, 2018-12-12:
I had a sensor smash a couple days ago but from me accidentally putting pressure on the glass 🤦‍♂️

We should make a good extruder mod that allows smooth pass through of a sharp edge filament 👍

AbeFM, 2018-12-12:
Iv'e seen a bowden-tube-with-notch-for-sensor that seems reasonable, and the indirect sensor seems reasonable (and protects sensor).

I guess I need to try designing one with a grooved pulley, I think it's much less of a blockage.

**It does seem it would hold back your testing to put it back in. Is it as simple as commenting a line of code? _Can you point me to where?*_*

AbeFM, 2018-12-13:
Not worth that much to me. Or more correctly, I couldn't keep it current and still support this development.

Considering putting it back on the Features Request list and we can talk about it from that perspective. :-)

Mjankor, 2019-01-04:
The "good extruder mod that allows smooth pass through of a sharp edge filament" has been done. You can feed anything you like into my indirect sensor modded extruder. You won't harm the sensor. The filament pathway is very constrained, and very smooth. 
https://www.thingiverse.com/thing:3223513

kmshort, 2019-01-30:
AbeFM. I'm about to do the extruder sensor mod myself, as posted above. If cutting works as well as you say it does, I'd be interested in a ZeroBeast firmware with cutting in it.

AbeFM, 2019-02-01:
I'm not involved with the project. I asked for an explanation of how it worked and it was never made clear. I would love to hear your experiences in detail, someday.

AbeFM, 2019-02-11:
I would like to keep this thread on topic - using the original blade to cut filament which doesn't load.

I still find filament misloads at times, and the MMU is hung at a point where I would otherwise find the cutter to have been used and the print to still be running. Still not pressing, but still true.

TheZeroBeast, 2019-08-20:
Still not a good feature to have IMHO.

closed 2019-08-20 by TheZeroBeast