TheZeroBeast/TZB-MMU2S-Firmware #46: Mark Rahmani

TheZeroBeast, 2018-12-12:
Robert McKenzie , I hit a weird bug. Had a second mmu needs attention. Rightfully so because filament (in position 4) had ran out. Put a new spool in, and it's trying to feed position 5 now. This is a 4 color print and I don't even have position 5 routed... It's possible I hit the wrong key when attempting to resume as I wasn't sure which does it (I guess the middle button does...) - tried other buttons to get back to position 4 but it wouldn't let me.

diomark, 2018-12-12:
Looks like you already opened it for me:) to add more details -

I reloaded a new spool but realized that the existing filament is half-way up the ptfe tube. Tried to push it out by pushing the new filament through, but it was also partially in the orange tube, but not far enough to grip.. Finally pulled old filament out (by pulling it over the bearings in the mmu using pliers, breaking it, and pulling both sections out).

From it's length I'm guessing mmu detected filament out, while printer must have still read that it had filament, but it got pulled which sounds like the right behavior..

Struggled to resume it (I wasn't sure if it was the middle button or a different button), and when it did resume, it went to position 5.. detected that it's empty, and failed again. I was tempted to put in filament at position 5 just to see what happens, but in the end just aborted the job.

TheZeroBeast, 2018-12-12:
Hey mate, yes, I beat you too it and made one seeing I was sitting down at the code.

I sent you a few messages on fb too if you wanted to talk on there more generally.

I would bazaars to say you might want to left the FINDA if it missed the runout, the print is checking it every half a second but if the ball doesn’t drop or is held up by a tinny price of filament it might trick it.
I have mine about in the middle of too tight on filament and not triggering to give an example.

If you hadn’t seen I did up a RC2 last night that I found an idler alignment issue in that will most likely help everything.

TheZeroBeast, 2018-12-12:
This was identified as spoolJoin being turned on for the record.

closed 2018-12-12 by TheZeroBeast