TheZeroBeast/TZB-MMU2S-Firmware #49: Preheat option to match Chris's pretty pla MMU - 160/60

diomark, 2018-12-12:
This belongs on the mk3's firmware side but you don't have issue tracking there.. Would love to see an option to pre-heat to 160extruder/60bed to match Chris's pretty PLA MMU v1.2 profiles.. his profiles start the temp at 160 to minimize oozing. There's no easy way to pre-heat to this from the printer itself.. (it defaults to 215/60 for pla..)

TheZeroBeast, 2018-12-12:
I’d prefer to keep it stock personally as I use it for different purposes.
Also, min extrude temp is 170c so no test extrusion could happen at this temp🤔

I can however track down where to modify the values in the FW so you can just download, change and compile 👍

TheZeroBeast, 2018-12-12:
Please modify lines **474-475** in Prusa MK3 Firmware file **configuration_prusa.h** which you would pull from configuration folder and rename to that for the mk3 config.

closed 2018-12-12 by TheZeroBeast