TheZeroBeast/TZB-MMU2S-Firmware #50: 215-257 Constantly Wrong

AbeFM, 2018-12-12:
2 issues so far:
1) When you start it up, it doesn't move selector, If y ou hit the middle button expecting to load filament "0", it instead ejects filament "1" a few inches into free space (missing selector).
This happens repeatably, after power off, etc.

If you move to a different channel, it homes.

Suspected cause: A "home invalid" flag is not being set/checked.

2) Going to the 5th position, then once more to the right puts you right into service mode, making it easy to accidentally overwrite bowden cal or erase eeprom. Is this on purpose?!

AbeFM, 2018-12-12:
Note - after one cycle of calibration, the MMU goes immediately back to regular mode, it does not stay in calibration mode.

TheZeroBeast, 2018-12-12:
Selector is working as designed, and stays put from before power off, for power panic support. Might be a middle button feed should rehome if it’s been manually moved since power off.

Service menu is on service position with middle button.
Very unlikely that someone gets into this menu without intent and then unlocks eeprom safety and erases.

AbeFM, 2018-12-12:
It's pretty handy for testing. I guess the only weird bit is how unexpected it was.

The misfeeding seemed to go away, I haven't seen it in a while. Perhaps it was from not enough resets afer flashing.

closed 2018-12-12 by AbeFM