TheZeroBeast/TZB-MMU2S-Firmware #54: MMU no more responding after several middle button rehome

adlamy, 2018-12-14:
MMU 2.1.5 & MK3 5.0.4 RC3

After 5 or 6 rehoming with middle button MMU lost connection with printer, after layer print, extruder goes on purge tower and stops, (all leds dead, no response from MMU buttons, idler turns free, no error message on LCD screen, pose and resume print from menu changed nothing and stop print was the same, filament stayed in bowden tube, unload filament didn't work. I had to press the cancel button on printer to reset.

I had the same issue with your previous firmware always after several rehoming with middle button.

TheZeroBeast, 2018-12-15:
Can you explain in more detail these several rehomes and also provide serial logs of the issue?
e.i. Rehome1: failed load @ change3
      Rehome2: failed unload @ change 23

adlamy, 2018-12-15:
sorry but I don't know how to record a serial log, but the issue occurs between fil. 1 and 5, printing with PVA support interface, PLA as fil. 1 and PVA as fil. 5 sliced with last Slic3rPE.
It was load fails on filament 5 due to dirty tips after one hour printing on a 5h print.
Every time it's appened, I cut the bad tip, but I had to press several times on the middle button (4 or 5 times) to have the selector going to position 5 and reload filament.
During rehome the selctor stoped randomly in position 1, 2 or 6 far right, I pressed middle button until selector find the correct position.
I get the feeling that all these failed rehomes are in relation with the final fail.

TheZeroBeast, 2018-12-26:
Fixes for this will be in full release for 2.1.5 once testing is finalised.

closed 2018-12-26 by TheZeroBeast