TheZeroBeast/TZB-MMU2S-Firmware #55: Printer fails to continue after purge lines

madxbgt, 2018-12-15:
Printer fails to continue after loading filament and printing the first purge lines on the build plate started from RC 3 and RC 4, flashing back to pre RC fixed the issue, then reflashed to RC 4 again issue remains.
Log file attached
[Recv.pdf](https://github.com/TheZeroBeast/MM-control-01/files/2682435/Recv.pdf)
[image]
[image]

TheZeroBeast, 2018-12-16:
Hey mate,

Could you please try the following before I update the Releases page?
Hopefully it'll fix this issue for you and anyone else that might come by it.

This is the MK3 side, you would use the MMU2 Latest TZB Release.
[MK3-5.0.4-TZB-BN2328.hex.zip](https://github.com/TheZeroBeast/MM-control-01/files/2683434/MK3-5.0.4-TZB-BN2328.hex.zip)

madxbgt, 2018-12-16:
Ok I used build 2328 with MMU 228 but the printer no longer detects that the MMU is connected

madxbgt, 2018-12-17:
Sorry mate misunderstood your directions, used MK3 2328 with RC4 259 MMU and that has fixed the issue

TheZeroBeast, 2018-12-17:
Nice 👍 good to hear

closed 2018-12-17 by TheZeroBeast