TheZeroBeast/TZB-MMU2S-Firmware #57: Print Stalls, with no error or indication.

RJ-Make, 2018-12-15:
Tried a 2 color ABS print and it was doing well, then at 28% it just stopped with no error or notification.  No response from all 3 MMU buttons

Not sure if this is a MMU/Printer issue or a Octoprint Issue..

Here is a screen shot of OctoPrint terminal window.. Recv: echo:busy: processing repeats for ever and ever and ever.. :-)

[image]
[image]

RJ-Make, 2018-12-15:
PS: You can can't cancel the print either. OctoPrint just sits there "Cancelling..." ..

TheZeroBeast, 2018-12-16:
Hey RJ-Make & JettoBlack,

I also was hit with this last night, about 50 or so changes in.

I have make a few changes today and just did a 2h100change test print with no issues.

Would one of you be able to retry your gcode with it please?
Only MK3 changes 👍

[MK3-5.0.4-TZB-BN2328.hex.zip](https://github.com/TheZeroBeast/MM-control-01/files/2683431/MK3-5.0.4-TZB-BN2328.hex.zip)

TheZeroBeast, 2018-12-16:
This quite well be the same as #54

jettoblack, 2018-12-16:
Testing MK3-5.0.4-TZB-BN2328.hex.zip and just had another freeze occur, this time during the initial purge line. While switching filaments from 3 to 1, the MMU2 successfully loaded filament 1 up to the Bondtech gears, and was showing a solid green light indicating a successful load, but then everything was stuck. The MK3 LCD still says 3 > 1. I was able to stop the print from the MK3 LCD, but the MMU2 was not responding.

I did not wait long enough to see if the safety timers would kick in, nor do I have a serial log, sorry. I'll try to get one next time.

RJ-Make, 2018-12-16:
I had another failure today, same failure mode. Just wanted to note that in both cases it failed over the wipe tower..

TheZeroBeast, 2018-12-16:
Ok, I'll need logs and gcode please. If you're running Octoprint you can enable serial logging so you can grab a days worth or so, then just point me to the time the issue occurred.

jettoblack, 2018-12-17:
Just had another freeze early on in a print, just before unloading filament 1.  Filament 1 is still loaded.

MK3 LCD shows "1 > 2".

MMU2 has solid green light on Filament 1. The idler is in the neutral position. No response to any of the 3 buttons.

Octoprint serial.log and .gcode attached.  The serial.log was just turned on prior to this print, and ends just after this print froze.

Note: The serial.log has some messages like this:
2018-12-17 01:58:19,602 - Send: N105 T2*60
2018-12-17 01:58:19,604 - Not sending T2, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware
2018-12-17 01:58:19,827 - Recv: MK3 => MMU 'T2 @toolChange:2'

I think this was because I forgot to set the number of extruders in the Octoprint profile, however, I also think this message is not related to the problem. The Tn commands actually ARE being sent to the printer. Otherwise the filaments wouldn't have switched during the purge, and the MK3 LCD wouldn't show "1 > 2". So I think these messages should be ignored.

The log basically ends with:
2018-12-17 02:15:34,479 - Send: N4095 T1*3
2018-12-17 02:15:34,482 - Not sending T1, that tool doesn't exist according to the printer profile or was reported as invalid by the firmware
2018-12-17 02:15:36,477 - Recv: echo:busy: processing
[...]

No acknowledgement that the MMU received the T1 command...

[serial(1).log](https://github.com/TheZeroBeast/MM-control-01/files/2684283/serial.1.log)

[MK3 MMU2_Purge_Test_3A_PLA_0.25mm.gcode.txt](https://github.com/TheZeroBeast/MM-control-01/files/2684285/MK3.MMU2_Purge_Test_3A_PLA_0.25mm.gcode.txt)

TheZeroBeast, 2018-12-17:
Thanks mate, checking it out, I'll see what I can find out, this is definitely different to what I encountered last. I'll keep you posted, thank you for your assistance 👍

TheZeroBeast, 2018-12-17:
So it gets to the gcode T1 but never sends the command to the MMU2 from the logs.

Could you please try this, I moved the disable extruder to after the T command is sent.

[Firmware.ino.rambo.hex.zip](https://github.com/TheZeroBeast/MM-control-01/files/2684910/Firmware.ino.rambo.hex.zip)

TheZeroBeast, 2018-12-17:
I have a lot more comments on now to try track it down if you'd be so kind to get it to fail in the same way with serial log.

[MK3-5.0.4-TZB-BN2329.hex.zip](https://github.com/TheZeroBeast/MM-control-01/files/2687357/MK3-5.0.4-TZB-BN2329.hex.zip)

TheZeroBeast, 2018-12-17:
I just tried the above with a 25m 8changr print across ex 1, 3, 5 and had no issues at all.

Its interesting you guys keep getting this so regular.

- Comms is ruled out as we don’t get that far 🤔

[image]

zjohnson16, 2018-12-18:
Am having the same problem as well. Printer freezes in purge tower at random layers. The filament has been pulled to the top of the extruder body. MK3 LCD says 2>1 and it appears to freeze immediately before filament would retract through the Bowden. Also the extruder motor appears to be stuck with a high holding current (becomes very warm after sitting for a while). I'm not an expert on the PRUSA firmware or stepper motors, but I'm wondering if the freeze has to do with an extruder motor shutdown temp because it occurs at random times, and is apparently not replicable on ZeroBeast's printer.

TheZeroBeast, 2018-12-18:
Yes, that’s what stock does, hold with high current zjohnson16.

Jettoblack, that is great!!! The smoking gun per say. I’ll have to scrutinise your report but from the top of my head it shouldn’t ever be in state 2 when processing a cmd.

Get buck to you soon

TheZeroBeast, 2018-12-18:
Hey, I have found the culprit as mentioned above, I'm working on a solution 👍 thank you for the great log.

TheZeroBeast, 2018-12-18:
Ok, I have a quick solution for this that will get us printing. I'll update release page now.
[MK3-5.0.4-TZB-BN2330.hex.zip](https://github.com/TheZeroBeast/MM-control-01/files/2689461/MK3-5.0.4-TZB-BN2330.hex.zip)

AbeFM, 2018-12-18:
I have had the printer sitting on the purge block on various firmwares, just sitting there dead doing nothing. Generally I can wait for the timeout, but I've seen this case where it doesn't come back. I'll have to try the new stuff.

Aside:
I have noticed that a poor power connection can make certain motors not function.

It sounds weird, but I can have 15-20 non-homing resets, wiggle the cables, and then it works.

And I've had (on ALL firmware) issues where touching the cables (often doing something weird like changing filament mid print) makes a motor stop not function on the next usage. It's not the motor's wiring, I believe, but the board's. I feel SOMETIMES vibrations can get stuff that loose.

jettoblack, 2018-12-19:
Unfortunately I just had another stall using 2330, although it got much farther into the print.

Also unfortunately, I left the printer on all day, so the serial.log grew to a huge size with all the state messages.

If I'm reading this right the error looks the same as before:

It gets stuck in state 2 starting around here:
2018-12-18 15:08:50,443 - Recv: MMU Pre-Conf Payload,0x4F 4B  1  09B
2018-12-18 15:08:50,444 - Recv: MMU loop, state=1
2018-12-18 15:08:50,447 - Recv: MMU loop, state=2
2018-12-18 15:08:50,448 - Recv: MMU Pre-Conf Payload,0x4F 4B  1  09B
2018-12-18 15:08:50,451 - Recv: MMU loop, state=1
2018-12-18 15:08:50,452 - Recv: MMU loop, state=2
2018-12-18 15:08:50,455 - Recv: MMU Pre-Conf Payload,0x4F 4B  1  09B
2018-12-18 15:08:50,459 - Recv: MMU loop, state=1
2018-12-18 15:08:50,460 - Recv: MMU loop, state=2
2018-12-18 15:08:50,463 - Recv: MMU loop, state=2
2018-12-18 15:08:50,464 - Recv: MMU loop, state=2
2018-12-18 15:08:50,465 - Recv: MMU loop, state=2
2018-12-18 15:08:50,467 - Recv: MMU loop, state=2
2018-12-18 15:08:50,468 - Recv: MMU loop, state=2
[...]

The printer keeps on printing for the next 1 min 10 secs, but then hangs at the next MMU command (T1):

2018-12-18 15:10:00,065 - Send: N17695 T1*55
2018-12-18 15:10:00,069 - Recv: MMU loop, state=2
[repeats 8 times]
2018-12-18 15:10:00,078 - Recv: E-AXIS Disabled.
2018-12-18 15:10:00,080 - Recv: MMU loop, state=2
[repeats until reset...]

[serial(3).zip](https://github.com/TheZeroBeast/MM-control-01/files/2693016/serial.3.zip)

zjohnson16, 2018-12-19:
I had another stall as well 11 hours into a 14 hour print.

raavhimself, 2018-12-19:
I downloaded yesterday the BN2327 so maybe 2330 has fixed this, but in my stalls, it looks like it is the MMU2 filament motor the one that freezes (getting pretty hot). With the printing stalled, I can pause it and tell the printer to eject the filament. There's some movement in the extruder motor and the screen says it's ejecting the filament... forever, and then that's all. The filament is there, if you loose the festo it's free from the extruder but caught by the MMU2, but it doesn't move or let be moved.

RJ-Make, 2018-12-20:
I was coming here to say I was able to complete 2 MMU (2 color) prints back-to-back without any stalling on build 2330, but looks like I may have just gotten lucky..

AbeFM, 2018-12-21:
I have found extruder grinds (I'm back on maybe 27) from the MMU not pulling. I've also found filament just sitting at the top of the extruder, without the MMU pulling on it.

I'm a ways back, working on a better extruder, so it may not be relevant.

TheZeroBeast, 2018-12-21:
Ok, I haven't had time to work on this for a couple days but I am quite confident I have finished tuning the comms protocol.

For all having MMU2 commands stop midway please try the follow code, I'm going to do a good 10h test before Christmas if possible and if I don't find any issues I'll be pushing it to the releases page.

[MMU2-2.1.5-BN260.zip](https://github.com/TheZeroBeast/MM-control-01/files/2702911/MMU2-2.1.5-BN260.zip)

AbeFM, 2018-12-21:
It would be awesome if you could add more tags - and partially this is my Gintcompetence. I'll try "the new fix everything" version, then when I want to go back I have the hardest time!

Also, I need the source code so I can tweak the accels and gear-sensor distance or I can't test it. I'll try to run this now though.

TheZeroBeast, 2018-12-21:
Just saw I had forgotten to add the top line to the mmu2 hex, please add as per readme unless I’ve reposted by then.

Source is already synced to git for this.

What tags do you propose, I am definitely a minimalist so I don’t want to bloat them unnecessarily.

TheZeroBeast, 2018-12-21:
[MMU2-2.1.5-BN260.zip](https://github.com/TheZeroBeast/MM-control-01/files/2704561/MMU2-2.1.5-BN260.zip)

AbeFM, 2018-12-22:
Just something so it's easier to match a particular pair of FW's together, so if I want to go back to something I ran I can get both halves. :-)

jettoblack, 2018-12-22:
Hey @TheZeroBeast for trying that new MMU build BN260, should I still use the MK3 BN2330 linked above?

TheZeroBeast, 2018-12-22:
Nah man, MK3 code in zip as well 👍

jettoblack, 2018-12-22:
The good news is, using 260 / 2330, I got several hours into a print with no stall, so that may be fixed.

The bad news is, I found a new freeze issue that occurred when restarting after a failed load, which I'll put in as a new bug.

raavhimself, 2018-12-22:
It looks like I'm always a version behind, but 259/2330 also stalls. Six hours into a print, while unloading, the extruder motor worked alright, but the MMU2 filament motor did not want to move. Screen read 5>1, Paused the machine, tried to unload with the menu, screen read unloading filament 5. Machine freezed, Hard reset necessary to continue

TheZeroBeast, 2018-12-22:
> It looks like I'm always a version behind, but 259/2330 also stalls. Six hours into a print, while unloading, the extruder motor worked alright, but the MMU2 filament motor did not want to move. Screen read 5>1, Paused the machine, tried to unload with the menu, screen read unloading filament 5. Machine frozen, buttons not responding. Hard reset necessary to continue

Hey mate, that is the version with the issue, try loading the zip version above.

Panayiotis-git, 2018-12-22:
@TheZeroBeast, are your latest changes pushed to Git?
Are the following correct?
MK3 b.2330 = https://github.com/TheZeroBeast/Prusa-Firmware/commit/594c8f1c2d55d643521a8ee2ceeb3d44922002a2
MMU2 b.260 = https://github.com/TheZeroBeast/MM-control-01/commit/14ce556b51db4d597ebd89ace126a49e19ab25f3

raavhimself, 2018-12-22:
Hey mate, congratulations!! You are the best!!!
I have just finished my first MMU2 complete multicolor print. I'd hug and kiss you out of pure joy (but that's because we, the basques, are warm people).
[image]
Considering that I ordered my MMU2 in December 2017 and that I ordered it along with some soluble printing filament (that's the measure of how optimistic I was back then), having completed
 a multicolor print is something special. Thanks!!!
There are a few things I'd change, though, but probably they come inherited from the PRUSA firmware, but I'll comment about them anyway. But that'll be later, after I perform more tests.
Now, let's enjoy the moment. Thanks again!!!

raavhimself, 2018-12-25:
consistent results
[image]

TheZeroBeast, 2018-12-26:
Nice work, it’s going good for me also, these fixes will be released ASAP once testing is finalised.

closed 2018-12-26 by TheZeroBeast

TheZeroBeast, 2018-12-26:
> @TheZeroBeast, are your latest changes pushed to Git?
> Are the following correct?
> MK3 b.2330 = [TheZeroBeast/Prusa-Firmware@594c8f1](https://github.com/TheZeroBeast/Prusa-Firmware/commit/594c8f1c2d55d643521a8ee2ceeb3d44922002a2)
> MMU2 b.260 = [14ce556](https://github.com/TheZeroBeast/MM-control-01/commit/14ce556b51db4d597ebd89ace126a49e19ab25f3)

Hey mate, sorry I missed replying, they were and will be updated to final 2.1.5 release today hopefully.