TheZeroBeast/TZB-MMU2S-Firmware #59: Filament selector gets stuck

joeycerone, 2018-12-16:
Running MMU 2.1.4 & MK3 5.0.3

My selector seems to get stuck occasionally when switching between filaments. Seems like it must be running into some resistance but I'm not sure where. Rods and POM nut look good, I've removed the blade cutter. I've loosened and tightening everything evenly and not too tight. Is there any way to make this less sensitive?

madxbgt, 2018-12-16:
I found some ptfe superlube on the rods helped my issue alot with that problem , and reducing the MAX_SPEED_SEL to around 5000 in config.h helped also.

joeycerone, 2018-12-16:
Thank you!! I'll grab some superlube tomorrow. As far as reducing the speed, I'm pretty new to all this, I could modify that file but I don't know how to generate the hex. Is that something that you could help me out with or post here?

madxbgt, 2018-12-16:
I am unsure how to create the hex file, but I was able to upload an update to the MMU with the Arduino IDE software.

Panayiotis-git, 2018-12-16:
I suggest to not use any lubricant at these rods. It is going to be a mess.
It is enough to change the speed to 5000 or less. I've changed the parameter and it is working fine since the first release of the firmware.

The whole change filament process takes a lot of time. The gain of driving the selector to a high speed is measured in msecs. And it is also puts a lot of stress to the plastic components.

TheZeroBeast, 2018-12-16:
Yeah, I dropped it a bit on this release, might drop it more if it'll make it more accessible, like you side, it doesn't save much time going quick. So, 5000 is working good for yourself at the moment?

Panayiotis-git, 2018-12-16:
Yes, no problem.

TheZeroBeast, 2018-12-16:
Cool 👍

joeycerone, 2018-12-16:
I inspected the code a bit last night and saw that it ran at 3000 in stealth mode. I think... I changed my print to stealth mode and it finished without any issues.

joeycerone, 2018-12-16:
> I am unsure how to create the hex file, but I was able to upload an update to the MMU with the Arduino IDE software.

joeycerone, 2018-12-17:
> I am unsure how to create the hex file, but I was able to upload an update to the MMU with the Arduino IDE software.

Ok, I'm trying to use the arduino software now to do this. I downloaded all the source files but I'm not really sure what to do next. The instructions on the readme say something about "version.h.in" but I can't find that file anywhere.

Can you give me a quick walk through on how to recompile with Arduino? I've used it once before so I'm a little familiar with it. I got the prusa board installed ok.

Also using a mac, but I'm guessing that shouldn't matter.

TheZeroBeast, 2018-12-17:
Hey mate,

Not sure how you got into stealth mode as my firmware doesn’t have much of not any of the methods implemented for it but I’d try changing it to 5000, this is what I’ll be trailing as the new default.

To make a hex file you have to open it first in a text editor and add the following line to the top as per readme file.

; device = mm-control

joeycerone, 2018-12-17:
Interesting, I just went through the menu like usual. The MK3 is definitely in stealth mode, maybe I've just been lucky on the selector since then. Unfortunately I did have the selector stick overnight so I'd like to figure out how to reduce this and recompile it on my own.

I'm just trying to recompile your files and I'm getting this error:

Arduino: 1.8.8 (Mac OS X), Board: "Arduino/Genuino Zero (Programming Port)"

In file included from sketch/Buttons.cpp:9:0:
main.h:16:21: error: avr/wdt.h: No such file or directory
 #include <avr/wdt.h>
                     ^
compilation terminated.
exit status 1
avr/wdt.h: No such file or directory

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

AbeFM, 2018-12-17:
The board is NOT an Arduino Zero.
[image]

joeycerone, 2018-12-17:
Thanks Abe! Just figured this out and came back to post an update. Looks like it's verifying correctly now. As soon as this finishes printing I'll try adjusting the value and seeing if I can get it running!

closed 2018-12-17 by joeycerone

TheZeroBeast, 2018-12-17:
Nice work, yeah, that’s the bird 👍👌

Stealth in MK3 is fine, I thought we were talking about the mmu2 😂🤦‍♂️

joeycerone, 2018-12-17:
What is the stock value for the selector speed?

TheZeroBeast, 2018-12-17:
Hard to say, the speeds are done differently.
Mine is steps per sec and they use a delay amount between steps.