TheZeroBeast/TZB-MMU2S-Firmware #61: Revolver moves to wrong position for loading

danielegger1, 2018-12-16:
Since the newest version MK3-5.0.4-TZB-BN2327 and MMU2-2.1.5-BN259 I have massive loading issue because the revolver moves too far and the filament cannot be loaded.
It does home before each new "try" so the position must be correct, but I think the issue comes that the revoler believes the selector is on another place.
This has also happened with the selector to me, as I have had often the case that it loads filament 3 instead of 4 for example. And I also had the cases where the filament 4 was supposed to load but the mmu loaded filament 3 (led showed place 4), this happened also several times.

I am running just single color prints and I did like 20 today so I went back to the previous version as this issue is quite annoying.

So from my 20 prints today I have had several issues with wrong selector / revolver positions, have you changed something in the code what would explain such a issue?

I am quite sure that it is not a hardware issue, cleaned the unit today before starting and I had nearly no issues with the previous version. The only reason why I swapped to the newer version was that the selector had sometimes issues after rehoming and positioning (heard a strange sound from the motor in that cases) because it was waaay too fast, this is fixed in the newer version.

Allram, 2018-12-16:
Had the same problem running the 2327version on MK3 and 259version on MMU2.
The selector didnt quite know where it was, i solved it on current print by adding a bit of filament on the far left side, so that the selector crashes into that filament and thinks that is the wall of the MMU2. Printing great for a few hours now :) Maybe this is fixed with the slowload fw?

TheZeroBeast, 2018-12-16:
Hi Guys, I'm not seeing any of this on my prints, can you please get a video of what is happening and some further detail?
e.g.
* is it first load after turning/connecting to the printer?
* is the selector aligned with the light position?
* does a rehome occur before it attempts to load that filament if first load?

danielegger1, 2018-12-16:
I will install your fw tomorrow again and make some videos.
Very hard to describe 

My printer was running the hole days with small prints, so I tested a lot of fresh loads after print starts and I think it happens more often if I forgot to prepare a filament for loading, so I get a loading issue on the mmu and then it‘s hard to get it into the extruder, because the revoler does not press against that current filament position. I have now found out that revolver homing was much louder on previous fw than current - can it be, that the revolver get‘s a wrong home position because of the new logic? Did you use less current for stall guard homing?

danielegger1, 2018-12-16:
And just to clarify
I really like your firmware and much appreciateyour development. I am really disappointed from the progress of prusa on the mmu and your implement of the filament sensor was genius!
With the original version I got skipped layers all day long. With your fw i suddenly get some issue, but still never had issues that my printer starts without filament in it (like prusas version does)

danielegger1, 2018-12-17:
[image]

I think I am closer at finding my issue, at least one of those. Got the same issue now on the older firmware, just not that often.
The screws are similar since weeks now, but for some reason the filament slips now and I have to put more force at the screws than normal, not sure why and I am quite sure that this is bad to have that much force at the revolver, but time will tell if that is good or bad.

But anyway there is still some positioning issues and I will try to figure it out and will explain you more in detail. As I am running from octoprint I will be also able to send terminal logs in case of an issue.

AbeFM, 2018-12-17:
I get the same sense, drum movements are dependant on the filament being
loaded and one may get incorrect positioning of filament is cocked in there
making more resistance, or un-preloaded and the gap causes overshoot...
Maybe.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

danielegger1, 2018-12-18:
Sorry pressed mistakenly the close button.
I think the homing current is too low this causes mispositioning during homing.
I just tried to load a filament, it slipped and the mmu was not able to move the filament.
I pressed the middle button, rehoming occured and got the same error. Tried this 3 times without luck. Before the fourth try I helper the revolver and pressed it upwards against the spring so the revolver can move freely. Before the selector moved to the slot I released the mmu and this time it worked perfectly.
So my assumption is that the homing current is too low for the revolver and this causes the mmu to home before it is really home (so basically it stops at the last filament slot not on the end stop „bracket“)
I am quote sure that I need to release my spring screws, but still I believe this could be improved / fixed in the firmwar. I am not sure if I release the screws that the mmu will be able to pull the filament from the repkord box because my path is not as smooth as it is supposed to be.

So to sum that up:
You could try what happen when you screw in your spring screws and like in my example and try to load to nozzle several times from different slots.
I will do further tests by the end of this week, but just wanted to let you know what I found out meanwhile.

TheZeroBeast, 2018-12-18:
Could you please double check your idler isn’t rubbing on the case anywhere?

If it definitely isn’t I’ll see what I can do and get you to test.

danielegger1, 2018-12-18:
Idler, thats the name :D didn’t get that in mind and called it revolver.
Yes I will try to figure it where the additional friction occurs from homing and will tell you what I find out. Not sure if it is rubbing somewhere now as I have my unit running now since a while and I didnt had that issue two weeks ago, it also appears now in the old fw but less (where I think the homing is slightly different) and the original fw homing is quite aggressive for probably that reason. Maybe just the mount of the idler needs some oil.

danielegger1, 2018-12-21:
Hello 
Thanks for asking
I have recorded a video and found out that it is definitve a home issues
In the end of the video you can see that I lift the mmu top part upwards that it can home freely and after that it loads correctly. In the rest of the video you can see that the idler is on a wrong place and the filament cannot be loaded, maybe a higher current is necessary?
By the way to lift the top part does not always help it also sometimes gets a wrong home position. And as said before this issues startet about 2 weeks ago not sure why.
I have attached two videos

https://www.dropbox.com/s/mcm8mh59umair1o/IMG_1948.MOV?dl=0

https://www.dropbox.com/s/av7yhdqhhftwo85/IMG_1949.MOV?dl=0

TheZeroBeast, 2018-12-21:
Ah, nice videos 👍👌

We must have vastly different motor resistances.

I’ll try a couple tricks when I get a minute and send through for you to test 👍

danielegger1, 2018-12-21:
Awesome thanks!
So your firmware will be “bloody smart” then ;)
Looking forward for a testfirmware to get this thing back running great.

AbeFM, 2018-12-21:
On the second video, it looks like the drum... clicks at the end of its travel - like there's a soft stop and a hard stop? I would blame it on the tension springs, but obviously not with everything open.

I used to suspect that feeling was the wires getting caught behind the drum.

Check the screws tightening the drum (carefully, it's not hard to split the part). I have drum that doesn't slip but the stl needs a bit of work. https://www.thingiverse.com/thing:3282939

I have had fewer homing issues since I switched, and generally see an improvement on tightening screws.

danielegger1, 2018-12-22:
Abe, 

I checked the cables on the backside and they do not cause the issue.
TheZeroBeast, did you have had already a minute to look into the code?

TheZeroBeast, 2018-12-24:
Hey Danielegger1,

I've had a look this afternoon but you'll need to test of course 😁 I'll get a set of hex to you today 👍

TheZeroBeast, 2018-12-24:
This build isn't going to play nice with other MK3 builds but can you please let me know if your idler homes at boot without massive lost steps?

[MMU2-2.1.5-BN263.hex.zip](https://github.com/TheZeroBeast/MM-control-01/files/2706609/MMU2-2.1.5-BN263.hex.zip)

danielegger1, 2018-12-24:
Thanks for that firmware I have tested it today, but unfortunately it was never a firmware issue it was a stupid user issue - so shame on me!
[image]
The problem was caused by this part here, there is supposed to be a "straight" side, but not on mine anymore as this has been gone over time becasue the screws got loose.
Last time when I checked the screws they were "tigth" but not on the correct side, so they were able to slip while homing. So i loosend them and put the on the straight side and tightend them as much as possible without breaking the part.
[image]
I found it out when I installed the original prusa firmware and damn that thing is slow, didn`t remember that is it like 10 times slower! But with that slow speed I was able to find out that it slips already when moving to homing position, but with your fast fw I was not able to see that.

From homing point of view they are quite different and maybe you can let me know which one sounds better to you, as both are working now (by the way the attached BN263 fw is not recognized from the printer, so I am using the BN259 fw now).

So sorry for all that trouble, I have to try that now with a mmu print, but if loading works as in the videos I am quite sure also the prints will work without issues.
I have attached again two videos with previous fw (MMU2-2.1.5-BN259-SlowLoadVersion) and your attached one (MMU2-2.1.5-BN263)

BN259:
https://www.dropbox.com/s/44ph7ctrm76p8sx/IMG_1976.MOV?dl=0

BN263:
https://www.dropbox.com/s/11ouvxhxcbb7pqr/IMG_1975.MOV?dl=0

TheZeroBeast, 2018-12-26:
All good, I think that tweaks are going to be better for all anyway, we’ll have to wait and see.
Glad you worked it out, happy printing 😊👍

closed 2018-12-26 by TheZeroBeast