TheZeroBeast/TZB-MMU2S-Firmware #62: Cannot Calibrate Z Offset or do a Single MMU Print

TheLightSpeed, 2018-12-17:
After updating to the latest on both the MMU and the MK3 I cannot get it to run a Layer Calibration from the Menu on the printer. I also cannot get it to select a filament on a single filament print. I have instead had to select a MMU print and designate a extruder in Slic3r. When I run a Z Offset it just runs mesh leveling and heats. It asks me what Filament I want and regardless of which I choose it just sits on the Home Spot and sits and sits.

TheZeroBeast, 2018-12-17:
Hi TheLightSpeed, I just did one fine on the couple fixes I'm working on for another issue, I will advise when this is ready for you to see if that fixes your issue.

TheLightSpeed, 2018-12-17:
Okay I’m the meantime I reverted to stock. The first TZB was really good other than a little grinding. Being able to set the Bowden length was really nice. While I had the handoff almost perfect, the not being able to do a single print where it asks the filament I want or the z height calibration was kind of a deal breaker. 

When you update I’ll reload it all. Man stock is sloooooow.

TheZeroBeast, 2018-12-23:
Next release not long off, maybe Boxing Day if time permits, give the fast lane another go and let me know if you have any issues.

closed 2018-12-23 by TheZeroBeast