TheZeroBeast/TZB-MMU2S-Firmware #63: Can't be compiled with Arduino

raavhimself, 2018-12-17:
It cannot be compiled. Gives back an error.
I'd use your hex files but I have changed the distance from the filament sensor to the bondtech gears so I had to build my own hex files.
It didn't compile with Arduino, Travis or MM-build-env-Win64. Then I tried your original files but it didn't work either. Then I looked for help from someone who really knows about programming. He says it's not my great ignorance. There's something wrong with the files you uploaded and they cannot be compiled

The changes I made are minor:

1- file MM-control-01/main.cpp, line 519
where it reads " moveSmooth(AX_PUL, STEPS_MK3FSensor_To_Bondtech, 385,false, false); " I wrote "moveSmooth(AX_PUL, STEPS_MK3FSensor_To_Bondtech, 722,false, false); "

2- file MM-control-01/MM-control-01/motion.cpp, line 21
where it reads " /2/ { 300, 330, 300},", I wrote" /2/ { 563, 619, 563},"

TheZeroBeast, 2018-12-17:
Hey mate, I just double checked and it’s compiling fine on my setup. I’m only using the latest Arduino IDE build. Ensuring the board MMU2 is selected.

What error does it give?

raavhimself, 2018-12-17:
Thank you very much. The board had not been selected.I already got the hex and I'm eager to start testing. Thank's again. You're the best!

closed 2018-12-17 by raavhimself

raavhimself, 2018-12-17:
AbeFM - No idea what the smooth movement does. I wish I did. I just did a search for STEPS_MK3FSensor_To_Bondtech and used logic and my distance multiplier (30mm/16mm). No idea if I did right or wrong.

AbeFM, 2018-12-17:
That's the same thing that happened to me. I remember not using it but I forget why. :-)

I believe you can just change the first one, which is meant as a configuration.
Again-> A definition would be nice here!