TheZeroBeast/TZB-MMU2S-Firmware #64: Question: Under-extruding after change

AbeFM, 2018-12-17:
More of a question:
How do I more carefully tune the changes? If I get the grinding gone, I find I've got big gaps in my priming tower, and the first layer is often weak.

It seems like tweaking Extra Reload Distance is the answer here, I've spent a lot of time juggling sensor-bondtech, parking position and extra reload.

I'm not sure what I'm asking. :-) I would like to have a better feeling on how to get grinding gone while still being able to print small amounts in a layer successfully. I will say I have more problems on the initial loads than on subsequent layers.

TheZeroBeast, 2018-12-17:
Yep, got you sorted. With the fsensor to bondtech just make sure it just reaches.

I have just been doing some tests with adding in some at Eritrea extrude on load as you said. And it has removed me last line that isn’t primed normally on stock settings or mine.

I’m useing -11 now which is working very well without extruder loosing steps. Loosing steps would be the limit, where you’re still pushing fastish and it can’t extrude. We’re just getting the load closer to the nozzle

AbeFM, 2018-12-17:
that all sounds great till -11. I'm playing between -2 and -3

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2018-12-21:
I've been getting weak first layers - things like hole outlines will be missing parts.

What's weird is I'll get a skirt - though not complete, it's more like there's a retract using the wrong reference point.