TheZeroBeast/TZB-MMU2S-Firmware #65: Use of Tx and Tc seems to stall/confuse the loading process.

AnatomicFlack, 2018-12-17:
I'm using Tx and Tc during single filament prints and I'm getting an odd behavior when doing so.  

When I use them, Tx before the bed and nozzle are fully preheated, and Tc, after bed and nozzle are up to temp and bed level, has been completed.  The Idler sits in a sort of limbo state and the MMU becomes completely unresponsive after feeding the filament to the sensor waiting for the MK3 to pull it into the extruder for initial load.  The result is that the idler bearing for the loaded filament just sits on top of the hobbed gear so it not moving out of the way and the filament has to force rotate the loading stepper because of the pressure on the gear from the bearing.  

When the print finishes if I leave everything as is when it's time for the unload the extruder unloads to the top of the body properly, but the MMU just sits there with "unloading filament x" on the LCD and never does anything and all buttons are unresponsive.  It requires a restart to get things back up and running again.  

I hope I've explained this correctly.  I'm not sure if this is because there is such a long time delay between heating/leveling and my issuing the second command.  Perhaps I'm hitting some sort of a timeout on loading when there is a gap between Tx and Tc.

AnatomicFlack, 2018-12-18:
That is correct as far as I understand Tx and Tc.  

This issue appears related as I believe they may be using the same commands: https://github.com/TheZeroBeast/MM-control-01/issues/62

TheZeroBeast, 2018-12-21:
Sorry mate 🤦‍♂️ Forgot to do the line at the top to be able to flash, have a look at the readme please. I’ll send a new copy with it done once I’m awake 👍

AbeFM, 2018-12-21:
What line? I'll give it a spin if I get clear idea of what versions of
everything to install. :-)

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2018-12-21:
[MMU2-2.1.5-BN260.zip](https://github.com/TheZeroBeast/MM-control-01/files/2704562/MMU2-2.1.5-BN260.zip)

AbeFM, 2018-12-21:
`// public variables:
int8_t filament_type[EXTRUDERS] = { 0, 0, 0, 0, 0};
const int filament_lookup_table[8][3] =
            /*0*/  /*1*/  /*0*/
    /*0*/ {{4600,   400, 2800},
    /*1*/  {2500,   100, 1500}, //Abe was 4000 -> 1500
    /*2*/  { 715,   755,  735}, //Abe longer sensor-gear distance, was 300
    /*3*/  {-610,  -610, -610},
    /*4*/  {6000, 10000, 6500},
    /*5*/  { 600,   300,  550},
    /*6*/  { 350,   200,  385},
    /*7*/  { 400,   200,  455}
};`

Can you stick that in?

AbeFM, 2018-12-21:
Looks like that should be:
```            /*0*/  /*1*/  /*2*/```
not
```            /*0*/  /*1*/  /*0*/```

AbeFM, 2018-12-22:
TIL if you flash the MMU first, it hangs the mk3 and you can't flash the mk3 till you downgrade the MMU so it stops messing with the mk3.,

On older builds, I can bypass it with a rod in the sensor. :-)

TheZeroBeast, 2018-12-22:
Yep, that’s understandable given the comms changes

AnatomicFlack, 2018-12-22:
Thanks, flashed perfectly.

It's refreshing when it's not a [PEBKAC](https://en.wiktionary.org/wiki/PEBKAC) issue. 

I'll let you know how it goes after I run it through some prints.

AnatomicFlack, 2018-12-22:
Problem solved.  Test print with Tx/Tc ran as expected without issue.  I was also able to run the First Layer Calibration without issue.  So it would appear Tx and Tc work properly in this firmware.  Thank you.  

I wanted to tell you sooner... amazing job with this... I only hope that Prusa will wake up and realize you have the proper approach here.  Pairing an indirect filament sensor setup using Prusa's own sensor with your firmware approaches some pretty amazing levels of reliability!

TheZeroBeast, 2018-12-22:
Good to hear, I'll push it out to all after a bit further testing.

Thank you, appreciate your comments 😊

closed 2018-12-22 by TheZeroBeast