TheZeroBeast/TZB-MMU2S-Firmware #69: Buttons on MMU often mis-select

AbeFM, 2018-12-21:
The buttons will often do the function of a different button - it seems like, in general, they'll do their own function,no function, or the function of one button to the left.

I suspect it might have to do with the fact that the buttons are "virtual", a reading on an ADC and a resistor network as I understand it.

Could it be that the read time is too short, that too little averaging is being done?

I don't know what's causing it, but it's annoying when 4 times in a row the selector moves instead of loading, etc.

TheZeroBeast, 2018-12-22:
I haven't had this issue what so every AbeFM, yes, the three buttons are voltage divider circuits to a single ADC channel. You might have to test some ideas on your setup to see what helps. Debounce might need to be implemented for the ADC reading as you say. Although if you're getting it and myself and others aren't might be an idea to check those resistor values first, one might be off quite a bit.

TheZeroBeast, 2018-12-27:
Any update on this issue AbeFM.

closed 2018-12-28 by TheZeroBeast

AbeFM, 2018-12-29:
I would be surprised if the resistor values are wrong - someone's pick-and-place chose the wrong slot?

When you do an ADC read, there's a settling time that you call out. I would start there, raising that.

AbeFM, 2018-12-30:
4 positions work ok
#2 (second channel, "1") when you press the button, instead of loading like it does on the other channels, the orange light turns on and off.

Selecting "load all" from the menu loads filament 1, then goes to two, flashes the lights down the chain as if it had done the loading, with the drum moving while the selector stays still.

Loading filaments from menu works, except channel 2 which doesn't load.

`PAT9125_init:1
echo:enqueing "M701 E0"
echo:enqueing "M701 E1"
echo:enqueing "M701 E2"
echo:enqueing "M701 E3"
echo:enqueing "M701 E4"
gcode_M701 begin
MK3 => MMU 'L0'
echo:busy: processing
echo:busy: processing
echo:busy: processing
MMU => MK3 'ok'
gcode_M701 begin
MK3 => MMU 'L1'
MMU => MK3 'ok'
gcode_M701 begin
MK3 => MMU 'L2'
MMU => MK3 'ok'
gcode_M701 begin
MK3 => MMU 'L3'
MMU => MK3 'ok'
gcode_M701 begin
MK3 => MMU 'L4'
MMU => MK3 'ok'`

AbeFM, 2018-12-30:
Good news and bad news on this one:
The sensor up there is bad, it only lights if I wiggle the wire just right. That may well be causing the poor behavior.