TheZeroBeast/TZB-MMU2S-Firmware #71: Hang after failed load & "Press the knob to resume nozzle temperature."

jettoblack, 2018-12-22:
Using BN 260/2330 I experienced a hang when trying to recover from a failed load.

The filament got stuck in the bowden tube while loading, which the MMU recognized as a failed load, and started blinking red on that filament LED. The MK3 was initially waiting over the purge tower. As I was clearing the jam, the extruder moved to the corner of the bed and started cooling down the nozzle as a safety feature.

After clearing the jam and feeding the filament up to the MK3 sensor, the message on the MK3 LCD changed to "Press the knob to resume nozzle temperature. 202 / 0". So it looks like it correctly detected the filament at the sensor.

However at this point everything was frozen. Pressing the MK3 knob did nothing (tried short press, long press, scrolling the dial). Additionally the MMU2 buttons were non-responsive. This was using Octoprint, if that makes a difference. As a last ditch measure I tried to pause the print in Octoprint to see if I could unpause it, but it wouldn't pause.

Here's what was left in the Octoprint terminal, but I didn't have serial logging turned on this time.

Recv: echo:busy: processing
Recv: MMU not responding
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: fsensor_autoload_check_stop - autoload DISABLED
Recv: Recv: MK3 => MMU 'Filament seen at extruder'
Recv: echo:busy: processing
[Recv: echo:busy: processing repeats...]
Changing monitoring state from "Printing" to "Pausing" [this is where I tried to pause in Octoprint]
[Recv: echo:busy: processing repeats until I reset the printer]

TheZeroBeast, 2018-12-22:
Hey Jet-black, sounds like you would of needed to pull that filament back to before selector and hit the middle button to clear fault as per README.md for such situations. I've had a print going for the last 24h without issue so I haven't had a chance to do a timeout load fail test, I'll try when I have it free 👍

TheZeroBeast, 2018-12-23:
In the 27h print that just completed it failed to load twice and recovery was as it should be, temp comes back up, loads again and off it goes.

When you said you tried to manually push filament until optical triggers was that when the MMU2 was still flashing?

jettoblack, 2018-12-23:
> When you said you tried to manually push filament until optical triggers was that when the MMU2 was still flashing?

Probably, yes.

TheZeroBeast, 2018-12-26:
Hey mate, closing as I have tested fully, these updates we have been tested will be released once I get a new fsensor to finish testing.

closed 2018-12-26 by TheZeroBeast