TheZeroBeast/TZB-MMU2S-Firmware #72: Where is the source code for the MK3 firmware

YoungR0se, 2018-12-22:
Hi,
I'm looking for the source code for the MK3 version of firmware that goes with the ZeroBeast MMU2 firmware.  The HEX file is no good for me, because I need to edit the printer to allow for a taller Z-axis than normal.
Can anyone tell me where to find this code?
Fairfarren

TheZeroBeast, 2018-12-22:
https://github.com/TheZeroBeast/Prusa-Firmware

closed 2018-12-22 by TheZeroBeast

YoungR0se, 2018-12-23:
Ok thanks, only when I verify the build, Arduino gives back a whole A4 page of errors.  And yes, I did follow the README about environment preparation.  Any suggestions?

YoungR0se, 2018-12-23:
I am using the provided HEX file from https://github.com/TheZeroBeast/MM-control-01/releases form the time being, but I need to be able to edit the firmware because my printer's Z axis is 350mm not 210mm, which is only as high as the HEX is programmed to go...

TheZeroBeast, 2018-12-23:
I’ll double check it pushed properly.

You got the einsy board installed, selected and the MK3 config is copied into the main folder as configuration_prusa.h?

YoungR0se, 2018-12-23:
Yes, I have copied the EINSY board file from the Variants folder and renamed it Configuration_prusa

TheZeroBeast, 2018-12-24:
Are you running latest arduino ide?
What errors as PRUSA code has a whole bunch of warnings but it should still say it compiled at the bottom.

YoungR0se, 2018-12-25:
The program is Arduino 1.6.9.  Errors include 'mmu_state' was declared 'extern' and later 'static' [-fpermissive], uart2.h:24: error: initializing argument 1 of 'void uart2_txPayload(unsigned char*)' [-fpermissive] extern void uart2_txPayload(unsigned char payload[3]);, mmu.cpp:312: error: invalid conversion from 'char*' to 'unsigned char*' [-fpermissive] uart2_txPayload(tempFx);

TheZeroBeast, 2018-12-27:
How did you download the source code? Clone to zip? The errors shown above aren’t in my code.

YoungR0se, 2018-12-28:
I download a zip folder from the code page

TheZeroBeast, 2018-12-28:
That page was a snippet in time at the very start of development and isn’t updated. You would need to download to zip from main repository page (green button on right)

https://github.com/TheZeroBeast/Prusa-Firmware

YoungR0se, 2018-12-30:
Ok, I have done as recommended.  Changed 2 numbers in the code and flashed it to the printer.  It works that much, but the printer does not recognise that there is a MMU unit connected.  The menu shows loading options for a single head setup.

AbeFM, 2018-12-30:
Did to also flash matching MMU firmware?

Is there some way to pair them up? It always seems hard to find a matched
set. It almost seems like it would be better under a single revision
control.

I'm probably wrong about that, but it is hard to tell what goes together.
Maybe use the date for revising number?

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes