TheZeroBeast/TZB-MMU2S-Firmware #73: Selector doesn't move after Update

KappaMi, 2018-12-24:
I have the problem that after the latest update the selector can no longer move. With the previous firmware it works

TheZeroBeast, 2018-12-26:
Hey KappaMi,
2.1.5 is being released soon which solves this issue.
Please note that selector now never homes at boot due to power panic support.

KappaMi, 2018-12-26:
Thank you for your reply. This "Issue" is with the latest FW youve made.

TheZeroBeast, 2018-12-28:
Hey Mate, can you please review this issue in relation to 2.1.5 release yesterday?

KappaMi, 2018-12-28:
Yes I will do it in 3 hours Mate, thx for the good work :)

KappaMi, 2018-12-28:
The MMU only moves if i press a button on it. If I dont do it, it doesnt start

TheZeroBeast, 2018-12-28:
This is how it is meant to be, it shouldn’t fully home until you try do something.

The print starts fine I take it?

TheZeroBeast, 2019-01-02:
If this is still an issue please reopen.

Please try stock FW to compare the start function, I’m sure you’ll be convinced it is the same desired motions.

closed 2019-01-02 by TheZeroBeast