TheZeroBeast/TZB-MMU2S-Firmware #74: After idle timeout, selector makes crackling noise while 1st LED blinks red

jettoblack, 2018-12-26:
I'm mostly curious to see if anyone else has noticed this on FW 260 / 2330.

Before the idle timeout occurs, the selector motor makes a kind of "waterfall" noise which is annoying but bearable.

After the idle timeout occurs, the selector motor initially becomes quiet, but after a few minutes starts to emit a steadily increasing crackling noise, and the leftmost LED flashes red in sync with the noise. As time goes on, the noise and blinking will decrease, then increase again, until the selector motor is powered up and goes back to the normal waterfall noise.

I made a video: https://youtu.be/PBR0PyhX09s

Any ideas?

TheZeroBeast, 2018-12-26:
Thanks Jettoblack, I'll have a look into this.

Panayiotis-git, 2018-12-26:
I've noticed this also.
Then, if you gently push/twist the power and uart2 connectors on the MMU2, the amplitude of the noise changes.