TheZeroBeast/TZB-MMU2S-Firmware #75: Loading Issues latest FW

KappaMi, 2018-12-26:
If the selector is on position 1 and i want to load the filament to the nozzle the LCD shows "Unloading Filament 5". If i want to load Filament 5 to nozzle the LCD shows "Unloading Filament 1". There is no filament loaded at all. The message does not disappear either. It is not possible for me to load a filament.

TheZeroBeast, 2018-12-26:
Please provide release details when you say latest FW, do you mean RC4?

KappaMi, 2018-12-26:
MMU 2.1.5 & MK3 5.0.4 RC4

TheZeroBeast, 2018-12-28:
Please advise if still present in 2.1.5

KappaMi, 2018-12-28:
Hi Mate, my print will ready in 2 1/2h. I will check it and come back later. See ya :)

KappaMi, 2018-12-28:
If I now want to load a filament to the nozzle, I get the message that the 5th filament will be unloaded but then the material will be loaded correctly. To do this you first have to press a button on the MMU because it is not homed and then the wrong filament is loaded (skips a filament). I now have a print start and it is running.

KappaMi, 2018-12-28:
If the MMU now unloads the filament during printing, it just stops and I have to confirm the middle button to continue. Not always, but with every 3rd filament change approx.

TheZeroBeast, 2018-12-29:
Sounds like you haven’t gotten around to doing the Bowden calibration.

As per readme you need to calibrate to end oh Bowden tube when it’s detached from extruder end.

KappaMi, 2018-12-29:
Im only able to calibrate filament 1. If i press the right button the selector doesnt move to the next position

TheZeroBeast, 2018-12-29:
That is correct. It is entirely different to Stock. This is to get a ball park for the fsensor so we don't have to load slow the entire Bowden tube.

How'd you go getting it flush with the end of the tube?

AbeFM, 2018-12-30:
Standing feature request---> auto calibrate Bowden length, manually
calibrate sensor-gear distance.

In part because people won't mess it up. :-)

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

KappaMi, 2018-12-30:
@TheZeroBeast now its working fine, I recalibrated the first Filament again and now its doing good. Good Job my Friend and thank you for you hard work and thanks for replying. Wish you a happy new year! 🥇 :)

closed 2018-12-30 by KappaMi