TheZeroBeast/TZB-MMU2S-Firmware #77: No filament loading after first homing

Panayiotis-git, 2018-12-27:
If the MMU is not homed, the press of any button first triggers the homing procedure and then does the appropriate function.

In the case of the middle button press, after the initialization, the filament loading is skipped.
In mmctl.cpp (ln 32) instead of
`    else if (!digitalRead(A1)) {`
it should be
`    if (!digitalRead(A1)) {`

Keep the nice work!

TheZeroBeast, 2018-12-28:
I had noticed this and agree it makes more sense to do the action once homed like left and right. Pegged for next release.

AbeFM, 2018-12-30:
There's certainly a chance this is part of my button issue.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes