TheZeroBeast/TZB-MMU2S-Firmware #80: Issue Loading TPU with Indirect Sensor

AbeFM, 2018-12-27:
I noticed a weird issue - I was loading TPU (95a), and the sensor saw filament and mk3 began loading.... But there was filament squirted out between MMU body and selector carriage, out of the right side but in the razor blade's gap.

As the purge tower was printed, this overflow was used up.

I wonder if the filament is hitting the indirect sensor wheel, not pushing it out of the way, bunching up, then forcing through.

I've been wanting to make a slotted wheel, I may have to try this. I'm curious about others' thoughts.

TheZeroBeast, 2018-12-28:
Even with the laser sensor I have only been able to load without issue. Unload speeds are still set too high in Slic3r and the filament bunches at extruder.

I think you're correct about it not getting past your indirect sensor and then bunching at selector.

If you find a way to get it working please let me know 👍, maybe the TPU load speeds may help in config.h

TheZeroBeast, 2018-12-30:
I e made some progress this afternoon getting the unload post ramming to be synced, i.e. both mmu2 and MK3 unload together. This will be great for tpu

AbeFM, 2018-12-30:
Got a version I should try?

My finda (MMU) sensor died, it could be a week till I get it working but
I'll try to crack it open today.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2019-01-02:
Yeah mate, testing with PLA at the moment and the synced handoff is very nice. I fee it would be the nail in the coffin for TOU/PLA prints although I can’t test as new Slic3r doesn’t do type profiles correct at the moment. It says flex is always pla (default)

AbeFM, 2019-01-04:
How new? Certainly my slicing from a couple weeks back reported TPU correctly.

TheZeroBeast, 2019-01-04:
1.42.0-Alpha has type issues

https://github.com/prusa3d/Slic3r/issues/1543

AbeFM, 2019-01-04:
> prusa3d/Slic3r#1543 <https://github.com/prusa3d/Slic3r/issues/1543>

Maybe they are trying to fix the random speeds when you use TPU on the
earlier versions.

Just use the release slicer.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2019-01-05:
Could do but the new one is soooo nice 😂
I do see the Slic3r team are fixing some issues now they're back from holidays.

TheZeroBeast, 2019-01-13:
Please try out 2.1.6 with the synchronised unloading

https://github.com/TheZeroBeast/MM-control-01/releases

closed 2019-01-13 by TheZeroBeast