TheZeroBeast/TZB-MMU2S-Firmware #81: Feature Request: Peep notifications

danielegger1, 2018-12-29:
As I am not always on site by the printer I would like to enable somehow peep notifications in case of a loading / unloading issue so I will be directly informed in case of any issue.

TheZeroBeast, 2018-12-29:
The firmware already has two action commands you can put any command/script on in Octoprint.
I personally do telegram messages with them.

// action:m600
// action:mmuAttention

danielegger1, 2018-12-29:
Ah that’s a great and I also have the telegram app installed but I didn’t know that I can add commands in that plugin, do you have a short guide for that?

TheZeroBeast, 2018-12-29:
Lookup curl telegram message.

You need to find your URL for your bot as well as your chat_id. Add them to command with a message at the end.

To get the command working I tried on my Mac until it was sending correct then pasted into octoprint for the corresponding action.

danielegger1, 2018-12-30:
Ok I found the required informations, just not sure where I can insert the curl codes, do I have to put them in a script?
By the way thanks again for your awesome firmware
Run several mmu prints the last days without any missing layers. I have still some loading unloading issues (rarely) but they are appearing because of my stupid spool placement.
I really would like to see PR take your firmware and donate you a lot of money as prusa will not come to that stage in the next half year I suppose.

danielegger1, 2018-12-31:
Ok I figured it out how to send messages from octoprint, but I do not get the messages from the printer in case of an load / unload issue, only when the printer moves to the mmu issue position. So in my test the m600 command did not appear.

and I was also not able to catch the // action:mmuAttention, as I think the plugin needs to message without //
How did you catched the terminal messages?

[image]

Recv: MK3 => MMU 'T1 @toolChange:3'
-> MMU issue 1.test no filament was loaded
Recv: MMU => MK3 'waiting for filament @ MK3 Sensor'
Recv: fsensor_autoload_check_start - autoload ENABLED
-> MMU issue 2.test - never reach filament sensor
... some times passes
Recv: // action:mmuAttention
-> MMU issue 3.test worked, printer went to issue position

TheZeroBeast, 2019-01-02:
Hey mate,

the messages to octoprint are good, your commands in the image above should be a bash command that octoprint can run.

for example I use the following for // action:m600

`curl https://api.telegram.org/botMYBOTIDNUMBER/sendMessage -d chat_id="MYCHATID" -d text="Filament Runout. Please feed me HUMAN!"`

danielegger1, 2019-01-02:
It should be like that?

[image]

Sounds quite strange for me, did you tried to use it with the Action Command Plugin?
And thanks for the curl example, I got that already working. But still I did not get any // action: m600 messages from the printer in my test cases.

danielegger1, 2019-01-02:
I now got the Recv: // action:m600 from the printer, I thought this is in case of a loading / unloading issue but actually that appears after a filament runout ;)

But: still I cannot get the plugin to work correctly (and execute something in case of the action command) as I think the "space" is wrong on the message, as it should be 
"Recv: //action:m600"
instead of
"Recv: // action:m600"

Please see the documentation here:
https://plugins.octoprint.org/plugins/actioncommands/

danielegger1, 2019-01-02:
Got it working after some debugging.
The trick is to use "virtual" printer from octoprint to test the action command, the issue was the system function call so I call now directly a script, currently I am abusing another file for the test function but it works as expected.
I would really like to see two other action commands // action:loadingIssue // action:unloadIssue
And if you do not mind I would also like to get a // action:crashDetected in case of a crash occured ;)

[image]

closed 2019-01-02 by danielegger1

TheZeroBeast, 2019-01-02:
Yeah, m600 is for a runout the attention one is for an issue, ie. old/unload/ground filament.

The space does look strange but it is how it supposed to be and has been working for me with the same octoprint plugin for a month now.

More commands would be very nice, I agree.
I’ll try add the mentioned in on a future release.

The scripts you’re trying to run, are they bash scripts, I don’t see any extensions on them

danielegger1, 2019-01-02:
Yeah for some reason that plugin does not like the system commands from within octoprint so I suppose they are not implemented correctly in some cases.

My scripts are bash scripts yes with optional arguments, the extension is not necessary from my understand (anyway it’s working :D) but I am still learning linux programming. Currently I am calling the main script (webcam, named because of the first / original function) and from there I call different functions like python scripts, telegram messages etc. so I have a short and easy main script with some other files behind to make the code easier to read (for me).
I am more familiar with C#, Rapid, Arduino,...
I will try to start learning a bit the Prusa code so I will be able to adapt things for tests.