TheZeroBeast/TZB-MMU2S-Firmware #82: Skelestruder Blues

AbeFM, 2018-12-29:
I'm trying to use the Skelestruder (gear reduced extruder with an indirect filament sensor), now, and now the buttons don't work at all. Kinda up a creek on this one, Prusa, you and JLTX are all at someone else's mercy. 
I have a few issues:
**1) Filament doesn't unload completely from extruder before MMU pulls on it,** stripping filament and failing.
I'm wondering if the steps to unload filament is hard coded?

Some loads and unloads:
'''start
echo: 5.0.4-TZB-2330
echo: Last Updated: Dec 29 2018 11:21:36 | Author: (none, default config)
Compiled: Dec 29 2018
echo: Free Memory: 2010  PlannerBufferBytes: 1392
echo:Stored settings retrieved
adc_init
CrashDetect ENABLED!
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
FSensor ENABLED
echo:SD card ok
echo:busy: processing
echo:busy: processing
fsensor_autoload_check_start - autoload ENABLED

MMU => MK3 '215'
MMU => MK3 '260'
MMU version valid
MMU - ENABLED
PAT9125_RES_X=0
PAT9125_RES_Y=240
PAT9125_init:1
echo:enqueing "M701 E0"
gcode_M701 begin
MK3 => MMU 'L0'
echo:busy: processing
echo:busy: processing
echo:busy: processing
MMU => MK3 'ok'
MK3 => MMU 'U0'
MMU => MK3 'ok'
E-AXIS Disabled.
MK3 => MMU 'T0 @toolChange:1'
echo:busy: processing
echo:busy: processing
MMU => MK3 'waiting for filament @ MK3 Sensor'
fsensor_autoload_check_stop - autoload DISABLED

fsensor_autoload_check_start - autoload ENABLED

echo:busy: processing
fsensor_autoload_check_stop - autoload DISABLED

MK3 => MMU 'Filament seen at extruder'
MMU => MK3 'ok'
E-AXIS Enabled.
MK3 => MMU 'C0'
echo:busy: processing
MMU => MK3 'ok'
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: paused for user
echo:enqueing "M702 C"
echo:busy: processing
echo:busy: processing
MK3 => MMU 'U0'
fsensor_autoload_check_start - autoload ENABLED

echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
MMU => MK3 'ok'
echo:enqueing "M701 E0"
gcode_M701 begin
MK3 => MMU 'L0'
echo:busy: processing
MMU => MK3 'ok'
MK3 => MMU 'U0'
MMU => MK3 'ok'
E-AXIS Disabled.
MK3 => MMU 'T0 @toolChange:1'
echo:busy: processing
echo:busy: processing
MMU => MK3 'waiting for filament @ MK3 Sensor'
fsensor_autoload_check_stop - autoload DISABLED

fsensor_autoload_check_start - autoload ENABLED

echo:busy: processing
fsensor_autoload_check_stop - autoload DISABLED

MK3 => MMU 'Filament seen at extruder'
MMU => MK3 'ok'
E-AXIS Enabled.
MK3 => MMU 'C0'
echo:busy: processing
MMU => MK3 'ok'
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: paused for user
echo:busy: paused for user
E-AXIS Disabled.
MK3 => MMU 'E0'
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
MMU => MK3 'ok'
E-AXIS Enabled.
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
MK3 => MMU 'R0'
echo:busy: processing
echo:busy: processing
MMU => MK3 'ok'
echo:enqueing "M701 E0"
gcode_M701 begin
MK3 => MMU 'L0'
echo:busy: processing
MMU => MK3 'ok'
'''
As the extruder is being disabled right when the filament is getting stuck in the extruder, I fear that may be the cause.

**2) Buttons on front panel don't work**
Yeah. Pressing them does nothing. I have no idea why, the only MMU code I changed was the speed array thing. Changes on mk3, but that shouldn't matter??

3) I'd still like a way to set the sensor-gear distance, flash and guess and flash is tough, especially since the filament doesn't pause at the "gear location", so it's hard to tell where the handoff happens.

AbeFM, 2018-12-29:
4) When loading filament, nothing comes out of the nozzle. It appears to under-load.
This is weird since the tube got shorter and I believe I fixed steps/mm. I think #1 and #5 are related.

5) Z dimension shows up as "?", live Z doesn't appear to work, though it's hard to say with "?" being shown as the dimension.

AbeFM, 2018-12-29:
I get the same behavior if I compile untweaked 2330 code.

TheZeroBeast, 2018-12-29:
Hey,
G21 has always done that in PRUSA FW for me anyway, it may just be hard coded to mm.

The not desolating thing is fprint not enabled for the arduino compiler. Check readme on where to add that parameter.

To change load/unload distances for a print it’s Slic3r but for all the inbuilt functions you’d have to track down where and update accordingly.

The stepper disable is what PRUSA do at the same time in unload, motor is free to turn.
I do see malformed ends get snagged a bit and haven’t been thinking of a solution, stay tuned.

The button thing, are you holding buttons long enough to trigger? They aren’t on interrupts yet.

AbeFM, 2018-12-29:
A bunch of that stuff started working.

I have made the changes to the environment, everything's been working fine for a month there.

G21 doesn't work for you either? Whatever hard coding there was is somehow missing now.

The extruder should unload filament past itself, just to be sure, no?

Even with MMU unplugged I get this no such thing as mm's issue.

AbeFM, 2018-12-30:
The not feeding/loading/unloading was the steps/mm being wrong. Somehow after recalibrating XYZ, that now works. I shouldn't have to tweak the "-85"

AbeFM, 2018-12-30:
Reinstalling the IDE fixed this.

closed 2018-12-30 by AbeFM

Panayiotis-git, 2018-12-30:
For the "buttons not working" issue, perhaps you have issued a Tn command (from an abandoned print?). From this point the MMU unit will not respond to buttons as usual until you issue an unload command.