TheZeroBeast/TZB-MMU2S-Firmware #83: Feature Request

gizzburn, 2019-01-02:
Hi awesome job on the firmware, is there any chance you could incorporate the new mesh 9 point bed leveling code in to your firmware in the next update, I tried it on the stock firmware on the Prusa MMU2 Fb page and it was loads better, this would be a great addition to your firmware.

thanks

danielegger1, 2019-01-02:
Thumbs up on that.
As far as I understand the new change in the fw it will be 3x3 or 7x7 possible just the additional N7 value at the G80 code.

TheZeroBeast, 2019-01-02:
Yeah, for sure, could you point me to the GitHub in question?

gizzburn, 2019-01-02:
I think this is the link your asking for...

https://github.com/prusa3d/Prusa-Firmware/issues/1239?fbclid=IwAR0tLZ61_lKBObPWM6iRfjwvAsUM8DGCQPkq93tuAIz7x8BEYztbj8rXsek#issuecomment-450579217

TheZeroBeast, 2019-01-05:
I have merged and going to do a couple tests. This will be in 2.1.6

gizzburn, 2019-01-05:
Great cant wait for the file..

TheZeroBeast, 2019-01-13:
This has been adding in 2.1.6, feel free to give it a go.

https://github.com/TheZeroBeast/MM-control-01/releases

closed 2019-01-13 by TheZeroBeast