TheZeroBeast/TZB-MMU2S-Firmware #85: Firmware stuck

migzz0301, 2019-01-03:
Using firmware MK3 5.0.4 RC 2370 & MMU 2.1.5 RC 265... while printing 2 colors. Sometime in the print it failed to load or unload it prompts on LCD "MMU needs attention..." after resolving the issue with stuck filament, it now loads the filament correctly to the calibrated length but then issue arises when you press knob to resume print. "press knob to resume" it just stuck there and no way to resume nozzle temp. i suspected the filament sensor for not detecting the filament thats why it had failed to load or unload in the first place. but i tested it... while resetting the printer, it triggered. I tried printing 3x still the same issue.

TheZeroBeast, 2019-01-05:
Yes, I agree, I have been working on a solution for this over the paste day and think I have it. 2.1.6 will be out soon that fixes this case where Sensor fires in the 'needs attention' state then doesn't reset.

migzz0301, 2019-01-05:
Looking forward for this bug fix. Thanks sir.

TheZeroBeast, 2019-01-05:
You're welcome. The commits are going in with the fix but some new features are still to be fully tested so might be a good idea to wait a bit.

Mjankor, 2019-01-07:
Cool. I've been having something like this problem too. Just random stops, like the MMU has been reset or disconnected during the print.

ssvenn, 2019-01-12:
I had a random stop where the MMU became unresponsive and none of the buttons did anything even after manually pulling the partially loaded filament out. Maybe a sensor misfire. I tried resetting the MMU and could get it to respond but couldn't get the MK3 to resume printing. More information on the LCD or serial console about what is blocking it from resuming would be great. Normally it should be possible to recover from a MMU reset, right?

TheZeroBeast, 2019-01-12:
Yep, same bug mate. FSensor triggers without a reset.

Panayiotis-git, 2019-01-12:
I think that with the current firmware, the moment you reset the MMU unit you doom your print to failure. After that, there is no way to trigger the correct response sequence to the MK3.

What I've changed and it seems to work, is that I have added some instructions to the MK3 firmware, to detect the "STR" message that the MMU unit sends to the printer when it restarts.

In the mmu.cpp class, in the mmu_loop method, I added the following lines.
```
    uart2_txACK();      // Send  ACK Byte

    // vvv New code vvv
    // Handle MMU unit reset
    if ((tData1 == 'S') && (tData2 == 'T') && (tData3 == 'R') && mmu_state != -1) {
    	  printf_P(PSTR("MMU restart detected. state=%d, lastTxPayload=%s\n"), mmu_state, lastTxPayload);
    	  mmu_ready = true;
    	  mmu_state = 1;
    }
    // ^^^ New code ^^^

    if (mmu_state == -1) {
```

Now, if I reset the MMU unit while the printer is blocked waiting, the printer detects the "STR" message and resumes from the waiting state. I then have time to pause the print, fix the error condition and resume the printing.

TheZeroBeast, 2019-01-13:
Good Idea Panayiotis-git, I'll see what I can do for 2.1.7

This issue has been resolved in 2.1.6, feel free to give it a go.

https://github.com/TheZeroBeast/MM-control-01/releases

closed 2019-01-13 by TheZeroBeast

AbeFM, 2019-01-14:
OMG awesome, btw. I've always thought this "should be", but beyond silent
prayers I didn't contribute much. :-)