TheZeroBeast/TZB-MMU2S-Firmware #90: Idler homing very loud latest build 2.1.6

madxbgt, 2019-01-13:
Updated to latest builds 2.1.6, idler homing is very loud for me anyway, as in issue #52 previously

TheZeroBeast, 2019-01-13:
hey mate, is it doing the 3s long noises again?

TheZeroBeast, 2019-01-14:
Damn. cool if I make you a homing guinea pig? I can do some tweaks and get you to try them.
I was getting some infrequent false homes so I made some tweaks.

madxbgt, 2019-01-14:
Sure mate, can do, room temp is about 23C

TheZeroBeast, 2019-01-14:
[MMU2-2.1.6-BN280.hex.zip](https://github.com/TheZeroBeast/MM-control-01/files/2754981/MMU2-2.1.6-BN280.hex.zip)

Here we go, first shot. Think it's a good one 😁

Report if homing is still load as hells bells

VintageSini, 2019-01-14:
I have the same loud homing issue with build 2.1.6 and am also in a 23C environment. However I compile from source code because I am developing a hardware mod for using the MMU2 in a lack enclosure. My MMU2 sits vertically on top of my enclosure and uses a redesigned selector with a different orientation for the FINDA probe. As a result my Filament_Parking_Steps needs to be set to -750 instead of the default -610 in your source code. 

Therefore, the above pre-compiled hex won't work for my design. Can I ask what I would need to change in the source code to prevent the loud homing issue? Is it just motor currents, or is it more? Or if it is easier for you could you attach a pre-compiled hex with all three Filament_Parking_Steps values set at -750?

AbeFM, 2019-01-14:
...Because you're not using my cool gear-drum?
https://www.thingiverse.com/thing:3282939

Generally when I had that going on, the screws were loose.

TheZeroBeast, 2019-01-14:
No, most likely not drum @AbeFM, different homing in this build which has been working great for me in the tropics.

@VintageSini once @madxbgt has tested I'll push the source and you can add your mods.
No it isn't simply changing a current, it's a balance of current/speed/sensitivity/acceleration

AbeFM, 2019-01-14:
When you do, a clearly matched pair would be helpful. Since I have to do a whole bunch of mods to both pieces of code, I don't want to get into it only to find out what I have doesn't match.

TheZeroBeast, 2019-01-14:
@madxbgt sorry mate, forgot to add the  **; device = mm-control** @ the top of the file.

This will work now

[MMU2-2.1.6-BN280.zip](https://github.com/TheZeroBeast/MM-control-01/files/2757466/MMU2-2.1.6-BN280.zip)

TheZeroBeast, 2019-01-14:
@VintageSini nice setup very neat. Yes, I'll get it online as soon as @madxbgt homing is sorted

madxbgt, 2019-01-14:
Thanks @TheZeroBeast all good now, is nice and quiet now, other than the homing issue the latest FW has been working great, thanks for the hard work.

TheZeroBeast, 2019-01-14:
Sweet, I'll push a small fix in 8h when I'm done at work.

Good to hear @madxbgt glad others are having success. I'm still trying to tune tips, Jam detection is picking up when the don't load so print doesn't skip layers but it's a pain to have to go snip end and continue. Are you guys running more than 1 cooling move and work sort of speeds?

closed 2019-01-14 by TheZeroBeast

madxbgt, 2019-01-14:
Only had one on a couple of prints but running 2 cooling moves, and running 22mm/s on unloading speed and increased ramming to 3sec , and altered the smoothed the ramp setting but other than that the settings you have seem good I use eSun PLA+ since its about the only cheap decent filament we can seem to get here in Oz without paying $100 for shipping the good stuff.

TheZeroBeast, 2019-01-15:
Yeah, cool, I need to find some time to actually do some solid tip testing.

Yeah, eSun is usually what I use but I just got 20 rolls from 3DFillies, 20micron tolerances now and prints soooo goood. Touch cheaper than eSun in Aus too.

madxbgt, 2019-01-15:
Nice had one spool from 3DFillies and did perform well, I'll have to try some more from them if they are going for 20micron, I think the eSun isnt really adhering to 50micron real well now.

Panayiotis-git, 2019-01-15:
I compiled the latest sources (https://github.com/TheZeroBeast/MM-control-01/commit/6619c072066356e9493aa5c0f02d69d7b80db9db) and the idler homing is again silent!

VintageSini, 2019-01-15:
@TheZeroBeast , Thanks. Works great. Its after Christmas and I'm singing Silent Night!

TheZeroBeast, 2019-01-25:
@madxbgt can you please see if this is noisy for you on idler homing?
I still need to make the home a little harder than previous. I load TPU and homing stops working, thinks the TPU is the end stop.
[MMU2-2.1.8-RC2-BN302.hex.zip](https://github.com/TheZeroBeast/MM-control-01/files/2798265/MMU2-2.1.8-RC2-BN302.hex.zip)

VintageSini, 2019-01-26:
Robert,

I was copied on this email and if you would like me to test(I have TPU), please provide the tree number. I'm new to GitHub and still making sense of it's structure. I can see the commits, but have no idea of how to associate that with builds (ie RC2 Build Number 302) 

On Friday, January 25, 2019 TheZeroBeast <reply@reply.github.com> wrote:

@madxbgt can you please see if this is noisy for you on idler homing?
I still need to make the home a little harder than previous. I load TPU and homing stops working, thinks the TPU is the end stop.
MMU2-2.1.8-RC2-BN302.hex.zip

VintageSini, 2019-01-26:
I honestly can't tell you what shore TPU it is because I bought it on clearance at Microcenter and it had no labels on the spool. The box it came in was labeled by Inland as flexible filament and that's all. It is fairly flexible, but prints very well. However, I have not tried it through the MMU. Also, I'm the guy who compiles from source because my exclosure requires a custom value for the FilamentParkingSteps variable so I cannot test with just the RC2 hex. Otherwise, the last release with the modified idler homing has been working well for me aside from  some false triggers from the mk3 filament sensor due to shavings building up.
On Friday, January 25, 2019 TheZeroBeast <reply@reply.github.com> wrote:

@VintageSini sure can mate, what shore TPU do you have, I have eSun eFlex 88A and it’s proving a pain to reload and push PLA out.

I’ll get RC2 hex out soon you can try with tpu