TheZeroBeast/TZB-MMU2S-Firmware #91: Filament loading loop

rhsdat, 2019-01-13:
While i was trying to print a Herringbone planetary gear with 5 colours i finally went into a loading loop at Color 4 (white).  See the [Video](https://youtu.be/uJrAY-vWaCU) ) . I was using 2.1.6 TZB 7x7 with fresh bowden calibration. The filament was correctly loaded but at the beginning of the waste tower print it begins again to unload.

TheZeroBeast, 2019-01-13:
Sounds like it’s detecting jams.

Can you provide gcode please for investigation?

rhsdat, 2019-01-14:
Sure....

[bearing5_0.2mm_PLA_MK3MMU2.zip](https://github.com/TheZeroBeast/MM-control-01/files/2753725/bearing5_0.2mm_PLA_MK3MMU2.zip)

TheZeroBeast, 2019-01-14:
Ah, the smoking gun(Jam detection isn't turned off for change).
Thank you for sharing your gcode.
I had overlooked the fact that feedrate backup/recovery wasn't implemented until the new alpha Slic3r builds.
If you re-slice in alpha3 then Jam detection will disable when it is supposed to. I'll update release page to advise others of this requirement.

Allram, 2019-01-14:
Got the same problem with a single color print (did not set the MMU to disabled).
Got 3-4 "jam detections" around 1hour into the print. After that the printer finished without any other stops :)

Gcode attached, sliced in Slic3r-Alpha3.

[MK3_PLA support_rouleau_sacs_poub_pat_V3_.zip](https://github.com/TheZeroBeast/MM-control-01/files/2754212/MK3_PLA.support_rouleau_sacs_poub_pat_V3_.zip)

TheZeroBeast, 2019-01-14:
Hi @Allram, this is standard Jam detection working.
I see you slic3d with Alpha3 although single colour prints wouldn't be affected by what @rhsdat was having above.
If you're getting Jam detects falsely try re-adjusting bondtech idler tension, dust on FSensor.

@rhsdat I had to recover from a few Jam's myself this afternoon from bad tips not loading and worked as intended. Please re-open this issue if you still have it once Slic3d with Alpha3.

closed 2019-01-14 by TheZeroBeast