TheZeroBeast/TZB-MMU2S-Firmware #92: Load / Unload speed when using a geared extruder (BMG)

skohls, 2019-01-15:
I'm using the latest firmware-combo for my MMU2 and my MK3. Yesterday I got my new Bondtech upgrade kit for the MK3 installed - which uses 3:1 gears.
I've adapted the E-steps to 830 (I'm using an 1.8° pancake motor) to get things going for the most part. 
Unloading w/o MMU was way too slow and made really long strings - so I modified your MK3 firmware to speed this up by 3x as well (trippling the feedrated in the "unload_filament()" function worked. Setting "UNLOAD_FILAMENT_1 to 21000 in Configuration_prusa.h didn't do anthing useful it seems).

Now, when I use the MMU2, my problems start again.
Loading to the nozzle works, might be a bit too slow tho. But unloading the filament again is way too slow in the extruder and creates some nasty strings (about 25mm long). Those would be no fun on the next load.

Is there any way you might consider another adapted build for BMG 3:1 setups? Or could you point me to the spots I would need to edit so I could build the firmware myself?

Thanks!

skohls, 2019-01-15:
[image]

skohls, 2019-01-15:
Sorry, closed it by misclicking... =^_^=

AbeFM, 2019-01-17:
Are you also using the Skelestruder?

I'm in the same boat, overly slow loads/unloads makes it picky.

I got the strings better with settings but it's been consistently worse
than stock in that respect.

I'd love to have better numbers for loading/unloading and will try tweaking
the numbers provided by the creator.

Not sure what packages make a "set", so I'm kinda sitting on old versions.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

skohls, 2019-01-17:
I'm using the new Prusa upgrade kit from Bondtech.
The settings will probably make the difference. Chris Warkocki commented on Facebook that he is tinkering with the setup as well - so I'm hoping he'll create some new "Pretty PLA" profile soon.

TheZeroBeast, 2019-01-20:
Hi @skohls, 2.1.8-RC1 has faster type 0 unloads again which are also able to be tweaked in config.h

Please try and advise if you issue persists.

skohls, 2019-01-20:
I've tried the new builds but that resulted in a lot of grinding from the MMU and only a partial unload. Seemes the MMU starts to retract while the extruder still has a good grip on the filament.
Also loading to nozzle didn't always work (didn't go far enough) and unloading (from the menu again) seems to be way too slow in the extruder.
I moved back to the 2.1.6-280 build for the MMU and the 2.1.6-TZB77-2378 with my changes to unload speed.
This is now also with my 0.9° LDO stepper in place, e-steps is set to 1660 in my MK3 now.

I've attached the part I modified (It's not like I really know how this all works together and stuff... =^_^= )

[changes.zip](https://github.com/TheZeroBeast/MM-control-01/files/2776677/changes.zip)

TheZeroBeast, 2019-01-20:
@skohls 
First, did you install both MMU and MK3 code as unloads are done with different command in 2.1.8-RC1 than 2.1.6 and if you're not running the this on MK3 it wouldn't do the synced unload and still be in bondtech gears.
In distance how far is your park position from your bondtech gears? Much different to the stock?

the loading settings will need to be adjusted in Slic3r to one, reflect 2.1.8-RC1 MMU parameters and then have the correct changes so the load is to the nozzle as intended.

skohls, 2019-01-20:
Yes, I always installed matching pairs of firmware onto MK3 and MMU.
I don't really understand the second question - could you rephrase this for Dummies (me) please?
Is this the distance one has to retract the filament to get it out of the grip of the bondtech gears?

TheZeroBeast, 2019-01-20:
Good, nice work.

Ok, with 2.1.8-RC1 the filament is parked 33mm back from nozzle which the code then retracts 55mm with both MMU&MK3 at the same time, if this distance is more for BMG I'll need to increase this distance.

So even if you can find the distance from nozzle tip to the center of Bondtech gears, from here I can make a small change that will hopefully get it working for you.

skohls, 2019-01-20:
I can't measure it myself right now but from the CAD files it seems to be  about 91.3mm.
https://github.com/BondtechAB/Bondtech_Prusa_i3

TheZeroBeast, 2019-01-20:
@skohls I was having a look in prusa fw and unless you have reset eeprom to default once you made the e-step change in the fw then you wouldn't of made the change.

If you haven't already I suggest you try load the default HEX files for 2.1.8-RC1 and then run "M92 E840" which will update the fw for BMG 3:1

Please try a load/unload with this done, at 91.3mm it may just be enough already when doing correct steps/mm.

skohls, 2019-01-21:
I'm using the LDO 0.9° stepper so my e-steps are at 1660 currently. I had checked the extrusion of the BMG extruder with an 1.8° stepper and just doubled those e-steps when I swapped in toe 0.9° stepper.

I'll do as you suggest - but since the weekend is over it might take a bit longer. ;)

wieckowy, 2019-01-21:
Hey.
I have the following configuration:
+ MK3
+ MMU2 (the latest TheZeroBeast firmware)
+ Bondtech extruder (3:1) with an LDO (0.9°) motor.

E-steps set to 1660 (for Bondtech with LDO engine). I did calibration BOWDEN_LENGTH and FSensor to BONDTECH Steps.

I have the same problem with unloading filament. On video from 0:36: [https://www.youtube.com/watch?v=iQCajLaaxB8](https://www.youtube.com/watch?v=iQCajLaaxB8)

 The problem only occurs in the latest version of TheZeroBeast. The previous version is ok.

Best regards!

skohls, 2019-01-21:
I did some tests after re-checking the calibration steps as in your readme.
2.1.8 results in a tug-of-war between MK3 and MMU. You can see in the picture that the MK3 wins.
2.1.6 default files result in a successful unload but produce a really long string.
2.1.6 with the faster unload fix in the MK3 produce a significant shorter string.
This is all done with the "Load to Nozzle" and "Unload Filament" from the MK3 menu.
[image]
[image]
[image]

TheZeroBeast, 2019-01-21:
[2.1.8-RC2.zip](https://github.com/TheZeroBeast/MM-control-01/files/2780326/2.1.8-RC2.zip)

skohls, 2019-01-21:
The tip looks a bit better but there is actually more filament "left over" from the unload.

[image]
[image]

TheZeroBeast, 2019-01-21:
Looks a lot better.
What do you mean by filament left over from unload?

wieckowy, 2019-01-21:
It's still bad. Extruder rotates too slowly when unloading filament (MMU pulls the filament much faster than the extruder's shafts spin).

In the previous version firmware was ok.

skohls, 2019-01-21:
The length of filament in the open - I just removed the PTFE tube from the MMU unit, nothing else.
In the RC1 version of 2.1.8 it was about 17.5cm to the tip, in RC2 it was about 20.5cm.
So the tip was a bit better but there was about 3cm of filament left in the tube before the MMU gave up.

skohls, 2019-01-21:
The tube was only removed to show the length of filament left from the unload. I just have to push the right button to finish the unload into the MMU, regardless of the tube being connected or not.
So no, it doesn't unload any better with the tube connected or disconnected.

The unload starts ok in the extruder but then the MMU starts to pull a lot too early. A lot of grinding inside the MMU happens and when the extruder finally lets go of the filament the MMU leaves a lot of it out in the open - since it has done it's steps I guess (the grinding)...

TheZeroBeast, 2019-01-21:
@skohls, ok so issue is clearing the extruder isn't obeying E-Steps from M92 command when doing the unload, each side gets same feedrate and should be in sync.

I'll have a look why this might be again.

Grinding would mean the unload won't complete fully also.

TheZeroBeast, 2019-01-21:
Ok, so I'll need some help. Can someone run there eyes over the plan_buffer_line method in planner.cpp, it looks like speed/feedrate is correctly calculated with e-steps/mm and mm/s?

File planner.cpp
LN: 652 Start of method
`void plan_buffer_line(float x, float y, float z, const float &e, float feed_rate, const uint8_t &extruder)`

LN: 743 Target step position
`target[E_AXIS] = lround(e*cs.axis_steps_per_unit[E_AXIS]);`

LN: 800 Distance in steps derived with steps/mm and current/target e step position
`block->steps_e.wide = labs(target[E_AXIS]-position[E_AXIS]);`

LN: 946 Inverse second made of feedrate to ensure no div by zero error
`float inverse_second = feed_rate * inverse_millimeters;`

LN: 964 Inverse second(feedrate) used against steps
`block->nominal_rate = ceil(block->step_event_count.wide * inverse_second); // (step/sec) Always > 0`

LN: 1182 Looks like another div by zero protection.
`block->speed_factor = block->nominal_rate / block->nominal_speed;`

I get lost about there.
From what I can see it is considering the step/mm in mm/s(feedrate) calc and if it wasn't every extrusion would be out be a factor of 6. IE. Nothing would print.

Any ideas guys? I don't have any geared extruders to test anything either which makes it tricky.

TheZeroBeast, 2019-01-22:
@skohls can you see if the code change I suggested above gets your unloads same speed as mmu?

skohls, 2019-01-22:
I tried but this was a total failure - the extruder kept spinning (rather slowly) while the MMU grinded the filament down and then gave up.

TheZeroBeast, 2019-01-22:
So @skohls you tried the below and still same as the fw zip above?

> For a very basic fix specifically for you guys would be to multiply the feedrate in my mmu_unload_synced method on LN: 469 in mmu.cpp
make it like bellow
plan_buffer_line(current_position[X_AXIS], current_position[Y_AXIS], current_position[Z_AXIS], current_position[E_AXIS], _filament_type_speed*6, active_extruder);

skohls, 2019-01-22:
> So @skohls you tried the below and still same as the fw zip above?

Yes, that's what I tried but it was actually worse than the fw zip above. 

Since 2.1.6 is basically working mostly - what has changed so much between 2.1.6 and 2.1.8 to break this unload function?

TheZeroBeast, 2019-01-22:
Ok, well all that has changed is the speed of synced unload from 2.1.6 and 2.1.8

There must be a hard feedrate limit you're hitting with your 6:1 setup, to test this you can try slowing the unload on both sides by changing one line of code on MMUCode LN: 151 in config.h
`#define TYPE_0_UnloadSpeed                     950  //  S/S`

On 2.1.6 it was 445 s/s which translates to 1405mm/m on MK3, just for reference.

Given you're running 1660 s/mm and we're trying to push 50 mm/s in 2.1.8 your trying to make the stepper do 83,000 s/s which I think is outside the ability of the stepper and control software.
Direct drive 1:1 MK3 extruder does 14,000 s/s to do that speed which is all good.

AbeFM, 2019-01-23:
I'm not sure how much it relates, I switched to a Skelestruder (3 or 3.5:1
belt drive), which the creator runs really slowly on retracts (to control
tip diameter, he says) and I get massive stringing. I haven't tried the new
firmware since I can't get better than 30 unattended changes now.

I've been wanting to make retracts/unloading better.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

TheZeroBeast, 2019-01-23:
@skohls did you get a chance to try dropping your microstepping and e_steps?

skohls, 2019-01-23:
I was about to try but it seems I can't successfully compile an MMU firmware yet. Flashing failed. I'll have to look into that in more detail., will take some time.

TheZeroBeast, 2019-01-23:
Cool.

Did you open hex and add this line to the top?

; device = mm-control

skohls, 2019-01-23:
Yes I did. But to be honest I'd just try and lower the speed - or go back 2.1.6 and stay on that firmware since it will probably be possible to get it to work well with slic3r settings - even tho I had some problems with tools out of sync (MMU choosing the wrong filament) after some problems during the last print. And possibly try my luck with stock firmware - the MMU turns out to be a real headache...

skohls, 2019-01-25:
I finally managed to get the MMU firmware to compile and build correctly - still getting the hang of the whole IDE stuff...
It works well with the unloads when I set ste type 0 unload speed to 650.

TheLightSpeed, 2019-01-25:
I would like to see this reopened. I have the bondtech extruder and this exact issue is happening. Also some goofy print quality issues.

TheZeroBeast, 2019-01-25:
@TheLightSpeed this is not an issue but a result of the bmg, the speeds just can’t be reached, you hit a limit.
As mentioned you can reduce microstepping to get back to the ballpark of usable step speeds on this 8bit avr system.

AbeFM, 2019-01-25:
Do you have a rule of thumb for how many steps/second you can run with
stock microstepping?
My extruder is 3:1, I went with a 1.8* motor for just this reason.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

3d-gussner, 2019-01-25:
This may help to figure out what the atmega2560 and marlin are capable on a geared extruder with 1.8 and 0.9 dev stepper https://reprap.org/wiki/Step_rates

cavotechnology, 2019-02-03:
> I finally managed to get the MMU firmware to compile and build correctly - still getting the hang of the whole IDE stuff...
> It works well with the unloads when I set the type 0 unload speed from 950 to 650.

Please share me compiled firmware for mk3 and mmu2. Thanx

cavotechnology, 2019-02-05:
Please reopen this issue @skohls @TheZeroBeast 
I have the same problem with unloading filament. On video (@wieckowy ) from 0:36: https://www.youtube.com/watch?v=iQCajLaaxB8

Prusa I3 MK3
MMU 2
latest firmware 2.1.8 RC2
Bondtech BMG extruder (3:1)
LDO (0.9°) motor
e-steps: 1660

Thanx

skohls, 2019-02-05:
@cavotechnology
I'll take a look how to get the files on here. Got the MK3 FW with 16 microstepping for the extruder (needs e-steps set to 830) and the MMU FW with unload speed set to 655 working well together. At least in regards of loads/unloads.

> > I finally managed to get the MMU firmware to compile and build correctly - still getting the hang of the whole IDE stuff...
> > It works well with the unloads when I set the type 0 unload speed from 950 to 650.
> 
> Share @skohls me please compiled firmware for mk3 and mmu2. Thanx

skohls, 2019-02-05:
So, those are the firmwares I'm currently using. The MK3 firmware includes several changes in regards to the BMG extruder and the 0.9° stepper as outlined by @codiac2600. The MMU firmware has a reduced unload speed. And as I'm no expert - use with caution!
You need to set your e-steps to 830!

https://github.com/skohls/MM-control-01/blob/MMU-FSensor/MM-control-01/MMU2-2.1.8-RC2-BN304_loadspeed655.hex
https://github.com/skohls/Prusa-Firmware/blob/MK3-MMU-FSensor/Firmware/MK3-2.1.8-RC2-TZB77-BN2386_plusCW_MS16_ES830.hex

cavotechnology, 2019-02-07:
@skohls thank you, firmware working perfect!!!

skohls, 2019-02-07:
Glad it works. The credit goes to @TheZeroBeast and @codiac2600 and several others tho - I just mixed and matched. Tell me if you get the MMU working with good tips on the filament - I have currently problems with stringing and no time to research via trial & error ;)