TheZeroBeast/TZB-MMU2S-Firmware #93: Nothing seems to work for me.

TNDavid, 2019-01-17:
ZEROBEAST firmware. HELP!
I cannot get the MMU2 to work reliably, so just updated with the latest 2.1.6-TZB-2378 printer, and 2.1.6-278 MMU firmware. Trying to go through the calibrations as shown on the github website. However, nothing works as stated! First, there should be LCD guided setup. Nothing on the LCD. It also says to enter setup at boot, but no mention of how. Or enter setup using the middle button "while on service location with selector". Again, HOW?

I do get into setup the old way, pressing the "X" reset button on the panel, and immediately holding down the middle button until all leds turn red.

When I set the "Bowden Length", and save it, it only pulls the filament back to just behind the FINDA. Then if I go into setup again to set the Finda to Bondtech distance, the filament still in the carriage is bent over or sheared off!

A LOT seems to be amiss here. Any help?

ssvenn, 2019-01-19:
Sounds like you have a hardware issue, what kind of filament are you using? 

To get into the service menu I just let the MMU start up normally, then press the right button to move the selector all the way to the right and push the middle button (there are six positions but only five filaments, the 6th is the service position and the led will start flashing)

Then the rightmost and leftmost led will light green to show that you are in the service menu, press left once to get to the bowden lenght setup mode and press the middle button.

It you keep sending the filament in and out with the middle button it should stop in nearly the same position every time. If it doesn't you have a mechanical problem that need to be sorted out. Too much friction, low quality filament, loose or badly centered pulleys.. It can be a lot of things.

For me, my main problem was too much tension while printing PLA, it is very easy to overtighten the tension springs so the MMU grinds the filament.

TNDavid, 2019-01-20:
Using PLA.  I will try loosening the tension screws a bit.  I have reverted to the latest released firmware, mainly because the Zerobeast firmware is jack hammering the selector drum in homing. It would surely break something if I keep it, so will sit this out and wait on future releases to try it again.

TNDavid, 2019-01-20:
One question:  The read me on github still has incorrect information on getting into the setup menu, but a couple people have told me about hitting the right button to get to the sixth position. That worked, but I would like to know where this information is coming from? If the github read me is wrong, we need to know where to look for the latest info.

TheZeroBeast, 2019-01-20:
@TNDavid any luck with this, the GitHub README.md is all inline unless you're reading something that isn't up to date, what like still has incorrect instruction?

TNDavid, 2019-01-20:
For some reason the readme.md I had found had the wrong instruction for
getting into setup!  I just opened the one on githhub and it is the right
one, right button to position 6. Do not know how I found the wrong one
before.

TheZeroBeast, 2019-01-20:
All good, glad it's showing correct.
Re-open if you're still having a specific issue, please provide detailed description and logs if possible.

closed 2019-01-20 by TheZeroBeast